verilator/verilator #3677: Developers: V5 freeze/release plan

wsnyder, 2022-10-15:
Developers, thanks to all of you, Version 5 is nearing stable release!

Here's my thinking as to timing.  Feedback is welcome - e.g. do we need more time?

~Oct 22nd (in one week): All feature merges must be in by this date.  After this point only important fixes will be merged. This will start a short pre-release period to check designs and OS distributions for any issues.

~Oct 29th (in two weeks): Release.

gezalore, 2022-10-18:
@dpetrisko 

> and anecdotally, seem to simulate quite a bit faster than v4

If you have some slightly-less-than-than-anecdotal numbers at some point that you would be willing to share, I would be keen to see them. It is good to know that the decisions and optimizations generalise as expected though.

kbieganski, 2022-10-19:
Is it realistic to merge [this](https://github.com/verilator/verilator/pull/3599) in time for V5? Would be nice to have event controls in class methods supported for the big release. I don't mind either way, so no pressure.

wsnyder, 2022-10-29:
Verilator V5 is released, thanks all!

closed 2022-10-29 by wsnyder