verilator/verilator #3678: VCD dump contents incorrect

Hiperino, 2022-10-15:
Hi!

I came across a strange behaviour in verilator with the VCD dump file. The code below:
```
module bug_tb;

	localparam CLOCK_HALF_PERIOD = 5;
	localparam CLOCK_CYCLE = 10;

	logic 			rst;
	logic 			clk;
	logic 			varA;
	logic 			varB;
	logic 			varC;
	logic			varD;

	initial begin
		$dumpfile ("SimLog.vcd");
		$dumpvars;
	end

	always #CLOCK_HALF_PERIOD clk = ~clk;

	always begin

		rst = 1;
		clk = 0;
		varA = 0;
		varC = 0;
		varB = 0;
		varD = 0;

		#CLOCK_CYCLE;
		rst = 0;
		varB = 1;

		$display("rst: %d, varB: %d", rst, varB);

		#CLOCK_CYCLE;

		$finish;
	end

endmodule
```
Compiles and runs correctly, giveing the output:
```
%Warning: previous dump at t=0, requesting t=0, dump call ignored
rst: 0, varB: 1
- bug_tb.sv:37: Verilog $finish

```

But when reading the VCD dump file in GTKWave the values for `rst` and `varB` remain unchanged during the entire simulation, as can be seen in the attached PDF: 
[gtkwave_out.pdf](https://github.com/verilator/verilator/files/9792410/gtkwave_out.pdf)

If I comment any on the assignments, for example:
```
varA = 0;
// varC = 0;
varB = 0;
varD = 0;
```

The dump files is correct:
[gtkwave_out.pdf](https://github.com/verilator/verilator/files/9792416/gtkwave_out.pdf)

A complete example of the bug, along with build script and generated files is here:
https://github.com/Hiperino/Verilator_bug

I have tested this on Icarus Verilog and it gives correct results in the dump file each time.

Am I doing something wrong or is it a bug?

What 'verilator --version' are you using?  Did you try it with the git master version?

Verilator 5.001 devel rev v4.228-153-g038d57070 
Most recent git master as of 15.10.2022

What OS and distribution are you using?
Arch Linux, gcc version 12.2.0, no SystemC

May we assist you in trying to fix this in Verilator yourself?
Yes

wsnyder, 2022-10-15:
The warning was indicating some data is dropped, so the trace file is incomplete.  Basically tracing was being called twice when using timing causing the warning, fixed.

Hiperino, 2022-10-15:
The warning diapered, but I'm afraid the issue still remains, the example that I linked above still gives wrong results.

wsnyder, 2022-10-15:
@kbieganski the problem is the traceActivity vector is set at the bottom of a function, but due to the co_await's that is no longer correct, it needs to be set before each appropriate co_await.  Can you look into this please? Thanks.

Simplified:
```
VL_INLINE_OPT VlCoroutine Va___024root___eval_initial__TOP__0(Va___024root* vlSelf) {
    while (1U) {
        vlSelf->t__DOT__rst = 1U;
        co_await vlSelf->__VdlySched.delay(0xaU, "/svaha/wsnyder/d/a.v", 30);
        vlSelf->t__DOT__rst = 0U;
        co_await vlSelf->__VdlySched.delay(0xaU, "/svaha/wsnyder/d/a.v", 35);
        VL_FINISH_MT("/svaha/wsnyder/d/a.v", 38, "");
    }
    vlSelf->__Vm_traceActivity[2U] = 1U;
}

```

See test_regress/t/t_trace_timing1.v

wsnyder, 2022-10-16:
Suspect remaining problem is identical to #3576, so let's discuss further there.