verilator/verilator #3682: V3Dfg.h: non-packed has no 'width()'

flaviens, 2022-10-17:
Hi there!

I found this error when building some SystemVerilog with Verilator

```
%Error: Internal Error: ../src/internalerror_0.1/src/pickled.sv:69815:24: ../V3Dfg.h:381: non-packed has no 'width()'
69815 |    logic [width_p-1:0] mem [els_p-1:0];
      |                        ^~~
%Error: Internal Error: Aborting since under --debug
Aborted
```

## Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
I created a repo for you to reproduce the issue.

## What 'verilator --version' are you using?  Did you try it with the git master version?
```
Verilator 5.001 devel rev v4.228-166-gcb7b024e8
```

## What OS and distribution are you using?
Ubuntu 22.04

Thanks!

gezalore, 2022-10-17:
I cannot reproduce the error you are reporting, instead get:
```
%Error: Internal Error: ../src/internalerror_0.1/src/pickled.sv:75454:44: ../V3DfgDecomposition.cpp:449: Missing source on variable vertex
75454 |   wire [levels_lp:0][aligned_width_lp-1:0] addr;
```
which was fixed in 840e26b69, and your repo is now passing for me. Can you advise on any other detail to reproduce the error you see?

flaviens, 2022-10-17:
Now I cannot reproduce it either, so I'll close the issue and reopen it if it re-appears.
Thank you very much!

FlyGoat, 2022-10-21:
@gezalore I'm getting similar issue on `Verilator 5.001 devel rev v4.228-185-g7b07cf912` which should have `840e26b` committed.

Reproduced on https://github.com/riscv-mcu/e203_hbirdv2, with verilator command:
```
verilator --no-timing --debug --gdbbt -Wno-UNOPTFLAT -lint-only -Irtl/e203/core rtl/e203/general/*.v rtl/e203/core/*.v
```

Error report:
```
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'__024root'
%Error: Internal Error: rtl/e203/general/sirv_sim_ram.v:44:18: ../V3Dfg.h:381: non-packed has no 'width()'
   44 |     reg [DW-1:0] mem_r [0:DP-1];
      |                  ^~~~~
```

Tried to reduce a minimal case but it seems like `sirv_sim_ram` module itself won't trigger the issue.

Could you please take a look?
Thanks.

gezalore, 2022-10-21:
@FlyGoat Could you provide accurate reproduction instructions? That command line you posted yields missing module errors.

FlyGoat, 2022-10-21:
> @FlyGoat Could you provide accurate reproduction instructions? That command line you posted yields missing module errors.

er.... Tried exactly same command locally and it reproduced perfectly fine.... Just FYI I'm on macOS13 using zsh....

gezalore, 2022-10-21:
I get 
```
%Error: rtl/e203/core/e203_cpu.v:487:4: Cannot find file containing module: 'e203_subsys_nice_core'
```
which lives in rtl/e203/subsys in my checkout, which is not on the command line, anyhow managed to reproduce it, patch imminent, thanks for reporting.

FlyGoat, 2022-10-21:
> I get
> 
> ```
> %Error: rtl/e203/core/e203_cpu.v:487:4: Cannot find file containing module: 'e203_subsys_nice_core'
> ```
> 
> which lives in rtl/e203/subsys in my checkout, which is not on the command line, anyhow managed to reproduce it, patch imminent, thanks for reporting.

Ah apologise for it, I've got some local modifications in my tree.

Can confirm this commit work.

Thanks!

commit 8b0d7199 (2022-10-21): DFG: don't try to call DfgVertex::width() on arrays

In DFG DfgVertex::width() is only defined for vertices representing
packed values, which DfgVertex::hash() used to violate. The only
non-packed values at the moment are DfgVarArray, which is a
DfgVertexVar, which are handled specially anyway, so this is easy to
fix.

Fixes #3682
closed 2022-10-21 by gezalore