verilator/verilator #3684: Version MACROs

dpetrisko, 2022-10-17:
Thanks for taking the time to report this.

> What would you like added/supported?

As a maintainer of library which serves many users with different versions of Verilator, I would like a way to switch implementations based on version numbers. 

Specifically, we provide a clock generator which uses the standard #period/2 scheme: https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_test/bsg_nonsynth_clock_gen.v. Because v4 of Verilator did not support #delay, we also provide a DPI version of the clock generator which Verilator can drive.

For this, we currently have 
```
`ifdef VERILATOR
dpi_clock_gen
`else
delay_clock_gen
`endif
```

Since v5 of Verilator supports #delay, we would like this to look like:
```
`ifdef VERILATOR
`ifndef VERILATOR_V5
dpi_clock_gen
`else
delay_clock_gen
`else
delay_clock_gen
`endif
```

I suggest maybe:
VERILATOR_VERSION_MAJOR (5, 6, 7...)
VERILATOR_VERSION_MINOR (month of release? Not sure if there's an existing schema)
VERILATOR_COMMIT_ID

Apologies if some of these exist already, but I only found verilator3 in the docs: https://veripool.org/guide/latest/extensions.html.

If you dislike this scheme, a macro for VERILATOR_DELAY_SUPPORT would solve this problem, since it is the most disruptive (and exciting) change for v5.

> Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)

I'm not aware of commercial simulators providing this switch, but it's their loss :)

> May we assist you in trying to fix this in Verilator yourself?

Yes, if you point me to the correct place to add.

dpetrisko, 2022-10-17:
If we could get this into the v5 release, I think it could help many users in our situation. It won't block our adoption of v5, but it will block us from leveraging the additional support. Thanks for the help!

dpetrisko, 2022-10-17:
That's a great point, I didn't know that. If that's the case, VERILATOR_TIMING would be the more relevant info

wsnyder, 2022-10-17:
There is already a VERILATOR_VERSION_INTEGER in verilated_config.h (included by verilated.h).   This was added in 4.440 (~ 6 months ago), so preexists V5.

I believe that should be able to be used for what you describe (note you can use >= and <= in #if expressions).  Closing assuming so, if not will reopen.

dpetrisko, 2022-10-17:
Oh great thanks! @wsnyder Should I open another issue for a VERILATOR_TIMING macro? The switch needs to happen for --timing as well

wsnyder, 2022-10-17:
We don't have any defines currently being set in the verilated output.

There's already a VM_TIMING in the make file, is that sufficient? You can in make use that to add -DTIMING or whatever.

dpetrisko, 2022-10-17:
That is helpful. It would be nice to have in the Verilog automatically since as a library we don't control the user's build flow. However, we could put an assertion that either -DVERILATOR_TIMING or -DVERILATOR_NO_TIMING is defined by the user if a module that depends on it is there. It's just a little more burdensome because folks will have to update their build flows to upgrade to the new version

wsnyder, 2022-10-17:
I realize I misread the above, you are looking for something to check in Verilog, not in C++?

dpetrisko, 2022-10-17:
yep! Basically to know whether or not we can rely on #foo to work

wsnyder, 2022-10-17:
Is just defining VERILATOR_TIMING sufficient?   (I don't see need for VERILATOR_NO_TIMING as that was what all old versions were and they didn't have a define).

But not version number? - that's harder as Verilog doesn't support <= >= like C++ preprocessor does.

dpetrisko, 2022-10-17:
VERILATOR_TIMING would be sufficient

wsnyder, 2022-10-17:
Committed, see extensions section in the manual for details, notably this is global and not line-by-line as e.g. looking at /*verilator timing_off*/ at each given line separately would do.  (That can be improved later if becomes a problem, which I doubt it will be.)

closed 2022-10-17 by wsnyder