verilator/verilator #3688: In Debian sid, build verilator with systemc of c++17 standard fail in 'examples/cmake_tracing_sc'

hushanjushi, 2022-10-19:
I tried to build verilator with systemc of c++17 standard, I have changed the systemc build standard and 'configure.ac' of verilator, but I met this:
```shell
/home/test/verilator/examples/cmake_tracing_sc/build/CMakeFiles/example.dir/Vtop.dir/Vtop___024root__DepSet_h84412442__0.cpp: In function ‘void Vtop___024root___ico_sequent__TOP__0(Vtop___024root*)’:
/home/test/verilator/examples/cmake_tracing_sc/build/CMakeFiles/example.dir/Vtop.dir/Vtop___024root__DepSet_h84412442__0.cpp:45:69: error: cannot convert ‘sc_core::sc_in<sc_dt::sc_bv<70> >’ to ‘WDataInP’ {aka ‘const unsigned int*’}
   45 |     VL_ADD_W(3, __Vtemp_h6c29bbf3__0, __Vtemp_h7c1a13d8__0, vlSelf->in_wide);
      |                                                             ~~~~~~~~^~~~~~~
      |                                                                     |
      |                                                                     sc_core::sc_in<sc_dt::sc_bv<70> >
In file included from /home/test/verilator/include/verilated.h:947,
                 from /home/test/verilator/examples/cmake_tracing_sc/build/CMakeFiles/example.dir/Vtop.dir/Vtop___024root__DepSet_h84412442__0.cpp:5:
/home/test/verilator/include/verilated_funcs.h:963:49: note:   initializing argument 4 of ‘WData* VL_ADD_W(int, WDataOutP, WDataInP, WDataInP)’
  963 |                                  WDataInP const rwp) VL_MT_SAFE {
      |                                  ~~~~~~~~~~~~~~~^~~
/home/test/verilator/examples/cmake_tracing_sc/build/CMakeFiles/example.dir/Vtop.dir/Vtop___024root__DepSet_h84412442__0.cpp:1630:37: error: invalid operands of types ‘sc_core::sc_signal_inout_if<sc_dt::sc_bv<70> >*’ and ‘WData’ {aka ‘unsigned int’} to binary ‘operator^’
 1630 |     if ((1U & (vlSelf->out_wide[0U] ^ vlSelf->top__DOT____Vtogcov__out_wide[0U]))) {
      |                ~~~~~~~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                   |                                           |
      |                                   |                                           WData {aka unsigned int}
      |                                   sc_core::sc_signal_inout_if<sc_dt::sc_bv<70> >*
/home/test/verilator/examples/cmake_tracing_sc/build/CMakeFiles/example.dir/Vtop.dir/Vtop___024root__DepSet_h84412442__0.cpp:1634:20: error: invalid operands of types ‘unsigned int’ and ‘sc_core::sc_signal_inout_if<sc_dt::sc_bv<70> >*’ to binary ‘operator&’
 1634 |              | (1U & vlSelf->out_wide[0U]));
      |                 ~~ ^ ~~~~~~~~~~~~~~~~~~~~
      |                 |                       |
      |                 unsigned int            sc_core::sc_signal_inout_if<sc_dt::sc_bv<70> >*
/home/test/verilator/examples/cmake_tracing_sc/build/CMakeFiles/example.dir/Vtop.dir/Vtop___024root__DepSet_h84412442__0.cpp:1636:37: error: invalid operands of types ‘sc_core::sc_signal_inout_if<sc_dt::sc_bv<70> >*’ and ‘WData’ {aka ‘unsigned int’} to binary ‘operator^’
 1636 |     if ((2U & (vlSelf->out_wide[0U] ^ vlSelf->top__DOT____Vtogcov__out_wide[0U]))) {
      |                ~~~~~~~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                   |                                           |
      |                                   |                                           WData {aka unsigned int}
      |                                   sc_core::sc_signal_inout_if<sc_dt::sc_bv<70> >*
/home/test/verilator/examples/cmake_tracing_sc/build/CMakeFiles/example.dir/Vtop.dir/Vtop___024root__DepSet_h84412442__0.cpp:1640:20: error: invalid operands of types ‘unsigned int’ and ‘sc_core::sc_signal_inout_if<sc_dt::sc_bv<70> >*’ to binary ‘operator&’
 1640 |              | (2U & vlSelf->out_wide[0U]));
      |                 ~~ ^ ~~~~~~~~~~~~~~~~~~~~
      |                 |                       |
      |                 unsigned int            sc_core::sc_signal_inout_if<sc_dt::sc_bv<70> >*
/home/test/verilator/examples/cmake_tracing_sc/build/CMakeFiles/example.dir/Vtop.dir/Vtop___024root__DepSet_h84412442__0.cpp:1642:37: error: invalid operands of types ‘sc_core::sc_signal_inout_if<sc_dt::sc_bv<70> >*’ and ‘WData’ {aka ‘unsigned int’} to binary ‘operator^’
 1642 |     if ((4U & (vlSelf->out_wide[0U] ^ vlSelf->top__DOT____Vtogcov__out_wide[0U]))) {
      |                ~~~~~~~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                   |                                           |
      |                                   |                                           WData {aka unsigned int}
      |                                   sc_core::sc_signal_inout_if<sc_dt::sc_bv<70> >*
/home/test/verilator/examples/cmake_tracing_sc/build/CMakeFiles/example.dir/Vtop.dir/Vtop___024root__DepSet_h84412442__0.cpp:1646:20: error: invalid operands of types ‘unsigned int’ and ‘sc_core::sc_signal_inout_if<sc_dt::sc_bv<70> >*’ to binary ‘operator&’
 1646 |              | (4U & vlSelf->out_wide[0U]));
      |                 ~~ ^ ~~~~~~~~~~~~~~~~~~~~
      |                 |                       |
      |                 unsigned int            sc_core::sc_signal_inout_if<sc_dt::sc_bv<70> >*
```
I can't find the solution in google or existed issues. It disappeared in c++14 standard, but I think if Archlinux use c++17, is there anyone know how to solve it?

hushanjushi, 2022-10-19:
I have change the part of 'configure.ac' from
```shell
case "$(which lsb_release 2>&1 > /dev/null && lsb_release -d)" in
*Arch*Linux* | *Ubuntu*22.04*)
_MY_CXX_CHECK_SET(CFG_CXXFLAGS_STD_NEWEST,-std=gnu++17)
_MY_CXX_CHECK_SET(CFG_CXXFLAGS_STD_NEWEST,-std=c++17)
;;
esac
_MY_CXX_CHECK_SET(CFG_CXXFLAGS_STD_NEWEST,-std=gnu++14)
_MY_CXX_CHECK_SET(CFG_CXXFLAGS_STD_NEWEST,-std=c++14)
_MY_CXX_CHECK_SET(CFG_CXXFLAGS_STD_NEWEST,-std=gnu++11)
_MY_CXX_CHECK_SET(CFG_CXXFLAGS_STD_NEWEST,-std=c++11)
AC_SUBST(CFG_CXXFLAGS_STD_NEWEST)
```
to
```shell
case "$(which lsb_release 2>&1 > /dev/null && lsb_release -d)" in
*Arch*Linux* | *Ubuntu*22.04*)
_MY_CXX_CHECK_SET(CFG_CXXFLAGS_STD_NEWEST,-std=gnu++17)
_MY_CXX_CHECK_SET(CFG_CXXFLAGS_STD_NEWEST,-std=c++17)
;;
esac
_MY_CXX_CHECK_SET(CFG_CXXFLAGS_STD_NEWEST,-std=gnu++17)
_MY_CXX_CHECK_SET(CFG_CXXFLAGS_STD_NEWEST,-std=c++17)
_MY_CXX_CHECK_SET(CFG_CXXFLAGS_STD_NEWEST,-std=gnu++14)
_MY_CXX_CHECK_SET(CFG_CXXFLAGS_STD_NEWEST,-std=c++14)
_MY_CXX_CHECK_SET(CFG_CXXFLAGS_STD_NEWEST,-std=gnu++11)
_MY_CXX_CHECK_SET(CFG_CXXFLAGS_STD_NEWEST,-std=c++11)
AC_SUBST(CFG_CXXFLAGS_STD_NEWEST)
```
I think the difference between archlinux and debian sid is it, but there is still errors

wsnyder, 2022-10-19:
It looks like the sc examples have been accidentally disabled from regressions for a month or so, so this fails on every OS as far as I can tell.  I will look into it.

hushanjushi, 2022-10-19:
> It looks like the sc examples have been accidentally disabled from regressions for a month or so, so this fails on every OS as far as I can tell. I will look into it.

Yeah, I have tried Archlinux and it just seems that the package maintainer didn't test this situation and it failed, too.

wsnyder, 2022-10-19:
Offending commit is 47bce4157dacd3b177a20308fde8f70e3b0ca929

hushanjushi, 2022-10-19:
> It looks like the sc examples have been accidentally disabled from regressions for a month or so, so this fails on every OS as far as I can tell. I will look into it.

And it may be systemc symbol change, please see this [issue of debian package](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1013046).(GCC may fix the bug but produce symbol changed) Is verilator hard 
-coding?

wsnyder, 2022-10-19:
@gezalore can you please take a look?

```
setenv SYSTEMC_INCLUDE=/usr/include  # So example will run - I'll clean this up to properly run it
cd examples/make_tracing_sc
make run
```

I tried adding "if varp->isSc() return true;" to keep() in V3DfgVertices.h (and think that should be in there) but it is more complicated, thanks.

gezalore, 2022-10-21:
> It looks like the sc examples have been accidentally disabled from regressions for a month or so, so this fails on every OS as far as I can tell. I will look into it.

@wsnyder the regressions should probably be fixed still

commit 6a3ec178 (2022-10-20): DFG: Do not inline SystemC variables

The emitted SystemC types (e.g. sc_bv) are not interchangeable with
Verilator internal C++ types (e.g.: VlWide), so the variables themselves
are not interchangeable (but can be assigned to/from each other). We can
preserve correctness simply be not inlining any SystemC variables (i.e.:
don't simplify any 'sc = nonSc' or 'nonSc = sc' assignments). SystemC
types only appear at top level ports so this should have no significant
impact.

Fixes #3688