verilator/verilator #3704: Shortcuts of verilator

MartinSolk, 2022-10-24:
Hi,
i tried to understand converted system verilog code which i didn't write by myself.
[image]
[image]
[image]
[image]
In the definition of some logic variable, there are short cuts which i don't understand and can't find in the documentation:
__PVT__,__Vcellinp__, __VinpClk__, __Vclklast__, __Vchglast__, __Vcellout__
I didn't find an explanation on the internet, so i guess it is generated by verilator.

Greetings,
Martin

wsnyder, 2022-10-24:
The internals are not intended to be readable without understanding how Verilator internals work, so these signals are not documented.  If you want to know what they do you will need to search for the appropriate magic words in the src directory, and read the description of what that signal or pass (comments at the top of the given source file) does.

Closing as answered but you can follow up here if you need.

MartinSolk, 2022-10-27:
Hi,
thanks for the answer. I understand all of them in the code except for once VinpClk. I dont understand, why it is generated? It seems that it is used to generate an offset:
[image]
[image]
[image]
As you can see this converted block is always one eval loop behind, but i don't really see why. Does anybody in general know what have to be in a system verilog file, that it get an offset like that? Has it something to do with the timescale?
Greetings,
Martin

wsnyder, 2022-10-27:
Vinpclk was used for generated clocks. This is removed in V5.000. That you are asking about it suggests you are debugging some clocking issue, which v5 generally typically completely solves, so I would suggest try git master, or wait a week for v5 to be released.

MartinSolk, 2022-10-27:
Ok i will try v5 thanks a lot