verilator/verilator #3705: `$ungetc` character type width should not be 32

jeras, 2022-10-24:
Tested using Git head.

While the standard is not exact here, the first argument for `$ungetc` is a character and should be 8-bit wide and should not expect a 32-bit value.
This is the reported error:
```
%Warning-WIDTH: ../../hdl/tbn/uart/uart_model.sv:123:13: Operator FUNGETC expects 32 bits on the $fungetc character, but $fungetc character's VARREF 'c_rx' generates 8 bits.
                                                       : ... In instance uart_model_tb.uart
  123 |     fs_tx = $ungetc(c_rx, fp_rx);
      |             ^~~~~~~
                ... For warning description see https://verilator.org/warn/WIDTH?v=5.001
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
make: *** [Makefile:126: compile] Error 1
```

jeras, 2022-10-24:
After I modified the offending line to:
```
    fs_tx = $ungetc(32'(c_rx), fp_rx);
```
I got the next error:
```
$ make TOP=uart_model_tb
verilator --binary -O3 -j 0 --timescale 1ps/1ps --x-assign fast -Wno-INITIALDLY --assert --trace-fst --trace-structs   --top uart_model_tb ../../hdl/rtl/tcb_if.sv ../../hdl/rtl/tcb_pas.sv ../../hdl/rtl/tcb_arb.sv ../../hdl/rtl/tcb_dec.sv ../../hdl/rtl/tcb_reg.sv ../../hdl/rtl/tcb_err.sv ../../hdl/rtl/gpio/tcb_gpio.sv ../../hdl/rtl/uart/tcb_uart_ser.sv ../../hdl/rtl/uart/tcb_uart_des.sv ../../hdl/rtl/uart/tcb_uart_fifo.sv ../../hdl/rtl/uart/tcb_uart.sv ../../hdl/tbn/tcb_pkg.sv ../../hdl/tbn/tcb_mon.sv ../../hdl/tbn/tcb_sub.sv ../../hdl/tbn/tcb_man.sv ../../hdl/tbn/tcb_tb.sv ../../hdl/tbn/gpio/tcb_gpio_tb.sv ../../hdl/tbn/uart/uart_model.sv ../../hdl/tbn/uart/uart_model_tb.sv ../../hdl/tbn/uart/tcb_uart_tb.sv
make[1]: Entering directory '/home/izi/VLSI/rp32/submodules/tcb/sim/verilator/obj_dir'
ccache g++  -I.  -MMD -I/home/izi/VLSI/rp32/submodules/tcb/../verilator/include -I/home/izi/VLSI/rp32/submodules/tcb/../verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_TIME_CONTEXT  -fcoroutines -std=gnu++17 -Os -c -o verilated.o /home/izi/VLSI/rp32/submodules/tcb/../verilator/include/verilated.cpp
ccache g++  -I.  -MMD -I/home/izi/VLSI/rp32/submodules/tcb/../verilator/include -I/home/izi/VLSI/rp32/submodules/tcb/../verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_TIME_CONTEXT  -fcoroutines -std=gnu++17 -Os -c -o verilated_fst_c.o /home/izi/VLSI/rp32/submodules/tcb/../verilator/include/verilated_fst_c.cpp
ccache g++  -I.  -MMD -I/home/izi/VLSI/rp32/submodules/tcb/../verilator/include -I/home/izi/VLSI/rp32/submodules/tcb/../verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_TIME_CONTEXT  -fcoroutines -std=gnu++17 -Os -c -o verilated_timing.o /home/izi/VLSI/rp32/submodules/tcb/../verilator/include/verilated_timing.cpp
/usr/bin/perl /home/izi/VLSI/rp32/submodules/tcb/../verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vuart_model_tb.cpp Vuart_model_tb___024root__DepSet_h9a4a1198__0.cpp Vuart_model_tb___024root__DepSet_ha22a7191__0.cpp Vuart_model_tb__main.cpp Vuart_model_tb__Trace__0.cpp Vuart_model_tb___024root__Slow.cpp Vuart_model_tb___024root__DepSet_ha22a7191__0__Slow.cpp Vuart_model_tb__Syms.cpp Vuart_model_tb__Trace__0__Slow.cpp > Vuart_model_tb__ALL.cpp
echo "" > Vuart_model_tb__ALL.verilator_deplist.tmp
ccache g++  -I.  -MMD -I/home/izi/VLSI/rp32/submodules/tcb/../verilator/include -I/home/izi/VLSI/rp32/submodules/tcb/../verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_TIME_CONTEXT  -fcoroutines -std=gnu++17 -Os -c -o Vuart_model_tb__ALL.o Vuart_model_tb__ALL.cpp
In file included from Vuart_model_tb__ALL.cpp:7:
Vuart_model_tb__Trace__0.cpp: In function ‘void Vuart_model_tb___024root__trace_chg_sub_0(Vuart_model_tb___024root*, VerilatedTrace<VerilatedFst, VerilatedFstBuffer>::Buffer*)’:
Vuart_model_tb__Trace__0.cpp:32:38: error: cannot convert ‘VlEvent’ to ‘CData’ {aka ‘unsigned char’}
   32 |         bufp->chgBit(oldp+4,(vlSelf->uart_model_tb__DOT__uart__DOT__sample));
      |                             ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                      |
      |                                      VlEvent
In file included from /home/izi/VLSI/rp32/submodules/tcb/../verilator/include/verilated_fst_c.h:25,
                 from Vuart_model_tb.cpp:6,
                 from Vuart_model_tb__ALL.cpp:3:
/home/izi/VLSI/rp32/submodules/tcb/../verilator/include/verilated_trace.h:444:57: note:   initializing argument 2 of ‘void VerilatedTraceBuffer<T_Buffer>::chgBit(uint32_t*, CData) [with T_Buffer = VerilatedFstBuffer; uint32_t = unsigned int; CData = unsigned char]’
  444 |     VL_ATTR_ALWINLINE void chgBit(uint32_t* oldp, CData newval) {
      |                                                   ~~~~~~^~~~~~
In file included from Vuart_model_tb__ALL.cpp:11:
Vuart_model_tb__Trace__0__Slow.cpp: In function ‘void Vuart_model_tb___024root__trace_full_sub_0(Vuart_model_tb___024root*, VerilatedTrace<VerilatedFst, VerilatedFstBuffer>::Buffer*)’:
Vuart_model_tb__Trace__0__Slow.cpp:98:35: error: cannot convert ‘VlEvent’ to ‘CData’ {aka ‘unsigned char’}
   98 |     bufp->fullBit(oldp+5,(vlSelf->uart_model_tb__DOT__uart__DOT__sample));
      |                          ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                   |
      |                                   VlEvent
In file included from /home/izi/VLSI/rp32/submodules/tcb/../verilator/include/verilated_fst_c.h:25,
[... cut ...]  ../../hdl/rtl/tcb_err.sv ../../hdl/rtl/gpio/tcb_gpio.sv ../../hdl/rtl/uart/tcb_uart_ser.sv ../../hdl/rtl/uart/tcb_uart_des.sv ../../hdl/rtl/uart/tcb_uart_fifo.sv ../../hdl/rtl/uart/tcb_uart.sv ../../hdl/tbn/tcb_pkg.sv ../../hdl/tbn/tcb_mon.sv ../../hdl/tbn/tcb_sub.sv ../../hdl/tbn/tcb_man.sv ../../hdl/tbn/tcb_tb.sv ../../hdl/tbn/gpio/tcb_gpio_tb.sv ../../hdl/tbn/uart/uart_model.sv ../../hdl/tbn/uart/uart_model_tb.sv ../../hdl/tbn/uart/tcb_uart_tb.sv
make: *** [Makefile:126: compile] Error 2
```

wsnyder, 2022-10-25:
1. ungetc takes an integer. This is because fgetc returns an integer, and because the "C" ungetc takes an integer.
2. It's illegal to cast a lvalue (something to be assigned to). This should give a verilator error instead of runtime error.  I'll use this as a bug to get that fixed.

wsnyder, 2025-03-16:
#3224 might want to get fixed along with this, in both cases we need to check for where a lvalue is needed and handle appropriately (as-is, deposit bits, or error).