verilator/verilator #3709: Support cross hierarchy assignment

balanx, 2022-10-26:
I hope verilator can did the following example code. 
Main synthesizer have supported this feature. It's very convenient to insert some DFT code into design.

```Verilog

module  sub2 (
  input  clk, rst,
  input  d,
  output stat,
  output reg q
);

assign  stat = ~q;

always @(posedge clk)
if (rst)
    q<=0;
else
    q<=d;
  
endmodule

module  sub1 (
  input  d,
  output q
);

sub2 sub2(.d(d), .q(q));
  
endmodule

module  top (
  input  clk, rst,
  input  d,
  output q, stat
);

sub1 sub1(.d(d), .q(q));

assign  sub1.sub2.clk = clk;     //cross hierarchy assignment
assign  sub1.sub2.rst = rst;      //cross hierarchy assignment
assign  stat = sub1.sub2.stat;  //cross hierarchy assignment
  
endmodule

```

wsnyder, 2022-10-27:
Cross-hierarchy is supported, the problem is the assignment to what otherwise is an input-only.  Perhaps a simplification is to only allow these assignments when the input is otherwise not connected.

We'd welcome a pull request to support this.  It's unlikely to be worked on otherwise.

BTW it might be that "bind" would also satisfy your needs in a cleaner SystemVerilog way.

balanx, 2022-10-29:
Thanks your reply.  You are right!
Verilator compiled OK after adding wire definitions for clk/rst. 

```Verilog

module  sub2 (
  input  clk, rst,
  input  d,
  output stat,
  output reg q
);

assign  stat = ~q;

always @(posedge clk)
if (rst)
    q<=0;
else
    q<=d;
  
endmodule

module  sub1 (
  input  d,
  output q
);

wire clk, rst, stat;

sub2 sub2(.d(d), .q(q), .stat(stat), .clk(clk), .rst(rst) );
  
endmodule

module  top (
  input  clk, rst,
  input  d,
  output q, stat
);

sub1 sub1(.d(d), .q(q));

assign  sub1.clk = clk;
assign  sub1.rst = rst;
assign  stat = sub1.stat;
  
endmodule

```

balanx, 2022-11-04:
But, is "bind' synthesizable ?

balanx, 2022-11-04:
I have tried in vivado. Dotted is OK, but "bind“ is failed.

donlon, 2023-09-14:
@balanx  It seems that verilator works well if ignore the warnings (`-Wno-ASSIGNIN -Wno-PINMISSING`) you can try this as a workaround.

wsnyder, 2026-03-29:
As noted above this works with `-Wno-ASSIGNIN -Wno-PINMISSING`, confirmed true still on master.

closed 2026-03-29 by wsnyder