verilator/verilator #3716: Internal Error:  ../V3DfgPeephole.cpp:558: Invalid zero extend

atorregrosasmd, 2022-10-28:
Another internal error (latest master, `5c658f8cd5650305f7eced91f01e7e297c43f244`)

```verilog
module t (
   input clk
 );

  function automatic logic foo(logic bar);
    foo = '0;
  endfunction

  logic a, b, res;
  logic [0:0][1:0] array;
 
  assign b = 0;
  assign a = foo(b);
  assign res = array[a][a]; 

endmodule
```

Result:
```
%Error: Internal Error: t_peephole_extend.sv:15:25: ../V3DfgPeephole.cpp:558: Invalid zero extend
   15 |   assign res = array[a][a]; 
      |                         ^
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

gezalore, 2022-10-28:
Thanks, I will take a look asap.

gezalore, 2022-10-28:
Similar to the other one, an earlier pass created a slightly malformed expression that DFG is then unhappy about.

commit 54c4351c (2022-10-28): V3Const: Do not introduce redundant AstExtend

Fixes #3716
closed 2022-10-28 by gezalore