verilator/verilator #3717: ./t/t_timing_protect.pl is non-deterministic

gezalore, 2022-10-28:
Keep rerunning `./t/t_timing_protect.pl --no-skip-identical` and should observe the hashes change, probably something depends on a memory address. 

@kbieganski could you take a look?

wsnyder, 2022-10-28:
BTW don't try with debug as it might (I didn't check) be stable because we turn off address randomization.

kbieganski, 2022-10-28:
:( I'll look at it next week.

wsnyder, 2022-10-28:
Looking at the command line it just isn't setting a stable key, so by design it is random.  I'll add it shortly.

kbieganski, 2022-10-28:
Oh, okay. I was confused because I removed everything from the test module and it was still nondeterministic. Thanks, I'm not familiar with how the protect stuff works.

closed 2022-10-28 by wsnyder

gezalore, 2022-10-28:
> Oh, okay. I was confused because I removed everything from the test module and it was still nondeterministic. Thanks, I'm not familiar with how the protect stuff works.

Nor am I apparently, should have been able to figure that one out :)

wsnyder, 2022-10-28:
protect encrypts the signal/file names using a key.  By default the key is picked at runtime from a true-random source, it can't be hardcoded to any computable algorithm, otherwise anyone with sources could run a dictionary attack to decrypt the signal names.  So all tests should always be giving a --protect-key that is constant.