verilator/verilator #3718: Segmentation fault with new Verilator versions

flaviens, 2022-10-28:
Hi there!

I encountered a segfault with `Verilator 5.001 devel rev v4.228-185-g7b07cf912` on Ubuntu 22.04.
[To assist you in debugging, I created a repo here to reproduce it](https://github.com/flaviens/verisegf2).
I'm sorry that I could not easily narrow it down further.

Thanks!

flaviens, 2022-10-28:
Thank you for your response! Sure does that help?

```
%Warning-UNUSEDSIGNAL: src.sv:260439:14: Bits of signal are not used: '_0005_'[6:0]
                                       : ... In instance rocket_mem_top.i_chip_top.system.tile_prci_domain.tile_reset_domain.tile.dcache.data.data_arrays_0.data_arrays_0_ext.mem_0_7
-node: VAR 0x26ca9380 <e11288062> {c260439ao} u1=0x38d83a40 @dt=0x21afaf70@(G/w8)  _0005_ [VSTATIC]  WIRE
260439 |   wire [7:0] _0005_;
       |              ^~~~~~
%Warning-UNUSEDSIGNAL: src.sv:269474:9: Signal is not used: 'RW0_clk_t0'
                                      : ... In instance rocket_mem_top.i_chip_top.system.tile_prci_domain.tile_reset_domain.tile.frontend.icache.tag_array.tag_array_0_ext.mem_0_0
-node: VAR 0x28122900 <e11309003> {c269474aj} u1=0x38dde640 @dt=0xaf028f0@(G/w1)  RW0_clk_t0 INPUT [VSTATIC]  WIRE
269474 |   input RW0_clk_t0;
       |         ^~~~~~~~~~
%Warning-UNUSEDSIGNAL: src.sv:269183:15: Bits of signal are not used: '_004_'[22:0]
                                       : ... In instance rocket_mem_top.i_chip_top.system.tile_prci_domain.tile_reset_domain.tile.frontend.icache.tag_array.tag_array_0_ext.mem_0_0
-node: VAR 0x27cd6d80 <e11309013> {c269183ap} u1=0x38ddec80 @dt=0x5ef0b60@(G/w24)  _004_ [VSTATIC]  WIRE
269183 |   wire [23:0] _004_;
       |               ^~~~~
%Warning-UNUSEDSIGNAL: src.sv:269184:15: Bits of signal are not used: '_005_'[22:0]
                                       : ... In instance rocket_mem_top.i_chip_top.system.tile_prci_domain.tile_reset_domain.tile.frontend.icache.tag_array.tag_array_0_ext.mem_0_0
-node: VAR 0x27cd6f00 <e11309014> {c269184ap} u1=0x38dded20 @dt=0x5ef0b60@(G/w24)  _005_ [VSTATIC]  WIRE
269184 |   wire [23:0] _005_;
       |               ^~~~~
- V3AssertPre.cpp:208:assertPreAll: 
- V3Ast.cpp:1121:     Dumping obj_dir/Vsrc_014_assertpre.tree
- V3Assert.cpp:516:   assertAll: 
- V3Ast.cpp:1121:     Dumping obj_dir/Vsrc_015_assert.tree
- V3LinkLevel.cpp:142:wrapTop: 
- V3Const.cpp:3678:   constifyAllLint: 
- V3Ast.cpp:1121:     Dumping obj_dir/Vsrc_017_const.tree
- V3SplitVar.cpp:1257:splitVariable: 
- V3Inst.cpp:623:     dearrayAll: 
- V3LinkDot.cpp:3318: linkDotArrayed: 
- V3Begin.cpp:341:    debeginAll: 
- V3Ast.cpp:1121:     Dumping obj_dir/Vsrc_022_begin.tree
- V3Tristate.cpp:1818:tristateAll: 
- V3Ast.cpp:1121:     Dumping obj_dir/Vsrc_023_tristate.tree
- V3Unknown.cpp:518:  unknownAll: 
- V3Ast.cpp:1121:     Dumping obj_dir/Vsrc_024_unknown.tree
- V3DfgOptimizer.cpp:237:extract: 
- V3Ast.cpp:1121:     Dumping obj_dir/Vsrc_025_dfg-extract.tree
- V3DfgOptimizer.cpp:244:optimize: 
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'__024root'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'__024paramod__0241e0019554db24e403a85d4260c10a25441229a90__05cplusarg_reader'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'__024paramod__0248963f76a25f0d49dea91f7c0dc2f82113d759a55__05cplusarg_reader'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'ClockGroupAggregator_2'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'ClockGroup_2'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'FrontBus'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'PlusArgTimeout'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'TLMonitor'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'TLMonitor_1'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'TLMonitor_10'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'TLMonitor_11'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'TLMonitor_12'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'TLMonitor_13'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'TLMonitor_14'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'TLMonitor_15'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'TLMonitor_16'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'TLMonitor_17'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'TLMonitor_18'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'TLMonitor_19'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'TLMonitor_2'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'TLMonitor_20'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'TLMonitor_21'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'TLMonitor_22'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'TLMonitor_23'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'TLMonitor_24'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'TLMonitor_25'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'TLMonitor_26'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'TLMonitor_27'
[... 238 lines cut ...]
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'data_arrays_0'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'data_arrays_0_0'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'l2_tlb_ram_ext'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'tag_array_0'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'tag_array_ext'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'AsyncResetSynchronizerPrimitiveShiftReg_d3_i0'
- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'data_arrays_0_0_ext'
[... cut ...]
%Error: setarch --addr-no-randomize /home/user/prefix-cellift/bin/verilator_bin_dbg src.sv -CFLAGS -std=c++11\ -Wall\ -g\ -O0 -LDFLAGS -pthread\ -lutil -Wall -Wno-PINCONNECTEMPTY -Wno-fatal -Wno-BLKANDNBLK -Wno-REDEFMACRO -Wno-PINMISSING -Wno-DECLFILENAME -Wno-TIMESCALEMOD -Wno-VARHIDDEN --cc --debug
%Error: Command Failed setarch --addr-no-randomize verilator_bin_dbg src.sv -CFLAGS -std=c++11\ -Wall\ -g\ -O0 -LDFLAGS -pthread\ -lutil -Wall -Wno-PINCONNECTEMPTY -Wno-fatal -Wno-BLKANDNBLK -Wno-REDEFMACRO -Wno-PINMISSING -Wno-DECLFILENAME -Wno-TIMESCALEMOD -Wno-VARHIDDEN --cc --debug
```

wsnyder, 2022-10-28:
@gezalore -  Definately Geza-land ;)

BTW while there please fix

- V3DfgOptimizer.cpp:262:Applying DFG optimization to module'tag_array_ext'

to not print each module with normal --debug, and add a space after "module", thanks.

gezalore, 2022-10-28:
> Thank you for your response! Sure does that help?

It helps with assigning blame, squarely to me this time

gezalore, 2022-10-28:
The crash is actually in the code path trying to print an internal. I will push a fix shortly for the DFG error that it was actually trying to print.

commit 0675510e (2022-10-28): DFG: Fix attempted evaluation of constants wider than 32 bits

Fixes #3718
closed 2022-10-28 by gezalore