verilator/verilator #3719: No warning if unpacked array bound limit is exceeded

goekce, 2022-10-31:
alu_tb.sv:
```systemverilog
module alu_tb
    (input clk);

logic [31:0] mem [32'h8000_0000 : 32'h8000_0000 + 2**16];
//logic [31:0] mem [2**32];

initial begin
    $readmemh("my.hex", mem);
    for(int i=0; i<20; ++i)
        $display("%d: %h", i, mem[32'h8000_000+i]);
    $finish;
end
endmodule
```

my.hex:
```
@80000000
B7C0EDFE 9380D0EA 37E176FF 130161F5
```

`tb.cpp`:
<details><pre><code>
#define vIncFile3(x) #x
#define vIncFile2(x) vIncFile3(V##x.h)
#define vIncFile(x) vIncFile2(x)
#include vIncFile(TOP)

#define VTOP3(x) V##x
#define VTOP2(x) VTOP3(x)
#define VTOP VTOP2(TOP)

#include "verilated.h"

int main(int argc, char** argv) {
	VerilatedContext context;
	context.traceEverOn(true);  // Enable signal dump generation
	context.commandArgs(argc, argv);  // Forward the arguments to the Verilated model
	VTOP top(&context);  // Instantiate the model

	top.clk = 0;
	while (!context.gotFinish()) {
			context.timeInc(1);  // Increment the time
			top.clk = !top.clk;
			top.eval();  // Evaluate the model
	}
}
</code></pre></details>

Compiling & running
```sh
verilator \
        --top-module alu_tb \
        --assert \
        --cc \
        --exe \
        --build \
        --trace-fst \
        -j \
        -CFLAGS -DTOP=alu_tb \
 alu_tb.sv tb.cpp

obj_dir/Valu_tb --verbose
```
outputs:
```
%Error: my.hex:1: $readmem file address beyond bounds of array
Aborting...
zsh: IOT instruction (core dumped)  obj_dir/Valu_tb --verbose
```

The index `0x8000_0000` seems to cause the problem. If I reduce it to `0x800_0000` in both `my.hex` and `alu_tb.sv`, code works fine. I have the hunch that the array indexes are integers, so the range is -2^31 to 2^31-1. I would expect that Verilator outputs at least a warning during compilation.

What 'verilator --version' are you using?  Did you try it with the git master version?
Verilator 4.228 2022-10-01, no I did not try.

What OS and distribution are you using?
Arch Linux

May we assist you in trying to fix this in Verilator yourself?
Yes

wsnyder, 2022-10-31:
Correct, as you noted the index by the standard is an integer, so basically you are asking for a negative index per IEEE.  You are right that specifying a constant that converts to negative is perhaps confusing, we'd certainly consider to add a new warning for when a constant gets converted like this.

As to the SystemVerilog code I suspect you were trying to implement before you found this, an associative array is likely what you want.

goekce, 2022-11-01:
More precisely the standard wants at least 24 bits for unpacked arrays. LRM 2017 7.4.2:

> Implementations may limit the maximum size of an [unpacked] array, but they shall allow at least 16 777 216 (2^24) elements.

And for packed arrays (7.4.1):

> The maximum size of a packed array can be limited, but shall be at least 65 536 (2^16) bits.

I will leave the issue open then.

Thanks for the alternative solution :)