verilator/verilator #3720: Docker not building

wsnyder, 2022-10-31:
Docker hasn't been building for over a year, it seems due to docker requiring a subscription.

@wallento, perhaps you could look into if there is a way to still support it for free?  Thanks.

Otherwise maybe we should just discontinue it to avoid confusion?

jjones-github, 2022-11-09:
I suspect everyone here is running from a local git repo but I had to build a container for the small team that I am working with currently. The ubuntu image in `apt` and the docker container were old enough that I felt compelled to upgrade. Here is a container if anyone needs a later version in the interim [docker](https://hub.docker.com/r/justinjones/verilator). This was built manually not with any continuous build process.

catkira, 2023-02-28:
@jjones-github can You upgrade Your docker image to 5.006?

jjones-github, 2023-02-28:
> 

Yes, I'll start it now. Hopefully I don't hit any snags.

jjones-github, 2023-02-28:
> @jjones-github can You upgrade Your docker image to 5.006?

I created a new version v2.06 which has verilator v5.006. Below is my favorite command to enter the container.

`docker run --rm --hostname QiDev:v2.06 --workdir /work -ti -v ${PWD}:/work --user $(id -u):$(id -g) --entrypoint /bin/bash justinjones/qi-dev:v2.06`

catkira, 2023-03-01:
can you also create a pure verilator 5.006 docker? qi-dev has a lot of other stuff in it.

jjones-github, 2023-03-01:
> can you also create a pure verilator 5.006 docker? qi-dev has a lot of other stuff in it.

I built a 5.006 version at `justinjones/verilator:v5.006`. It is still pretty large, 2.14GB but less than the 2.93GB in the qi-dev:v2.06 image.

PeterMonsson, 2023-03-22:
@jjones-github Hi Justin, can I convince you to update to ubuntu 22.04 or something similar that uses GCC10 or higher?

My small playground with your excellent docker image fails when I insert a delay in the testbench. I have traced the problem to GCC9 that doesn't support the -fcoroutines option

https://github.com/PeterMonsson/verilator-github-action

catkira, 2023-03-22:
and while you are at it, it would be nice to add cocotb, pytest, pytest-parallel, cocotbext-axi :)

jjones-github, 2023-03-22:
I'll start another container build with 22.04 and the extra tools. @PeterMonsson if you want to try a container that has `clang version 14.0.0-1ubuntu1`, you can try [this one](https://hub.docker.com/repository/docker/justinjones/qi-dev/general). It is larger but is ready now and I would got for `v2.08`.

jjones-github, 2023-03-23:
So `v5.009` and `latest` have been updated to 22.04. Version `v5.009_coco` is being built that has the extra python libraries already installed; `cocotb`, `pytest`, pytest-parallel`, and cocotbext-axi`. Note that `v5.009_coco` is only an amd64 image. The coco libs don't work on arm64 currently.

PeterMonsson, 2023-03-23:
@jjones-github it works! Thank you. This is amazing

jlloyola, 2023-05-02:
I took a shot at creating a [github action to build and push verilator docker image](https://github.com/jlloyola/verilator/blob/docker_build/.github/workflows/docker.yml).
However, the build seems to timeout or get killed for exceeding compute resources (https://github.com/actions/runner-images/issues/6680). I tried reducing the number of cores used by the `make` command, without success.
Any clues on why the build is so resource hungry? Are the build resources required documented anywhere?

wsnyder, 2023-05-02:
I have not seen that problem before; the default .git action rules that do builds seem to fit within the normal resources.

jlloyola, 2023-05-03:
I traced the issue to Github runner flakiness. The image was built and pushed without a problem after a retry.

PR #4163 (2023-05-04): Github action to build and push docker images
This PR enables a github action to build and push verilator docker image when tags are pushed to the repository (fixes #3720).
The following secrets must be configured in the repository using [these steps](https://docs.github.com/en/rest/actions/secrets?apiVersion=2022-11-28#create-or-update-an-organization-secret):

- `DOCKER_HUB_NAMESPACE`: docker hub namespace (should be `verilator`).
- `DOCKER_HUB_USER`: user name for logging into docker hub (must be a member of the `verilator` namespace).
- `DOCKER_HUB_ACCESS_TOKEN`: docker hub access token. S [... cut]
closed 2023-05-04 by wsnyder