verilator/verilator #3723: Parameters overridden with `-G` are missing from FST waveform

jeras, 2022-11-03:
Using latest Git version.

If I override a parameter at compile time with the -G argument, the changed parameter does not appear in the FST waveform file.

wsnyder, 2022-11-03:
It's not obvious to me how this happens, V3LinkDot seems to simply replace the old parameter with the new one well before tracing, can you please make a test_regress format test that shows the problem, and ideally try to make a pull request to fix it?

jeras, 2022-11-03:
I can try to create a regression test, but I need some help finding an example to copy from.
I did a grep through the test regress folder for `-G` and found:
```
test_regress/t/t_flag_parameter_bad.pl:    v_flags2 => ['-GPARAM_THAT_DOES_NON_EXIST=1'],
test_regress/t/t_flag_parameter_hier.pl:    #verilator_flags2 => ['-GPARAM=10 -Gtop.t.x.HIER=20'],  # HIER would error
test_regress/t/t_flag_parameter_hier.pl:    verilator_flags2 => ['-GPARAM=10'],
test_regress/t/t_flag_parameter_pkg.pl:    verilator_flags2 => ['-GPARAM_A=1'],
test_regress/t/t_flag_parameter_pkg.pl:    ms_flags2 => ['-GPARAM_A=1'],
test_regress/t/t_lib_prot_clk_gated.pl:                "-GGATED_CLK=1",
test_regress/t/t_lib_prot_clk_gated.pl:                             "-GGATED_CLK=1",
test_regress/t/t_param_public.pl:        verilator_flags2 => ["-GTOP_PARAM=30 --exe $Self->{t_dir}/$Self->{name}.cpp"],
test_regress/t/t_savable_format1_bad.pl:    v_flags2 => ["--savable -GMODEL_WIDTH=40"],
```
Do I use `v_flags2` or `verilator_flags2`?
How do I check for the presence of a parameter in the FST file?
Is there a reference simulator supporting FST files I can use as reference?

wsnyder, 2022-11-03:
For example you could make t_flag_parameter_hier_fst, for the FST code look at e.g. t_trace_fst.pl - this will convert to vcd and make a .out file with human readable contents and the parameter should be viewable in there. Thanks.

jeras, 2022-11-06:
I checked if the overriden parameter is also missing from VCD files and it is.
Since a VCD test is easier, I used that one.

I had trouble modifying `t_flag_parameter_hier` for this test, so I restarted from `t_timing_trace` and created `t_timing_trace_param_override`. To create the reference VCD file, I just run the test without parameter override and used the VCD output as the golden reference for the test.

I created a pull request:
8d94314e658c0cf214f5c8369863e2cae62b0f32

closed 2022-11-11 by wsnyder