verilator/verilator #3724: Nested `generate for` inside `generate if` not executed

jeras, 2022-11-03:
Using latest Git main.

I tried to create a small unit test, but I was not able to repeat the issue with simple code.

I probably have to learn some more advanced debugging techniques for Verilator.

I also think I encountered many issues related to the use of `@(posedge clk)` inside `task`s inside `initial` statements that look like race conditions, so I had to add unit delays which I usually avoid as the plague. I will try to write some regression tests for those, but I do not understand that part of the SystemVerilog spec well enough to be sure what is the correct behavior.

So to reproduce the issue:
```bash
git clone https://github.com/jeras/TCB.git
cd TCB
cd sim/verilator
make TOP=tcb_vip_tb
```
The code not getting executed is here:
https://github.com/jeras/TCB/blob/main/hdl/tbn/vip/tcb_vip_sub.sv#L129-L141
The `tcb.DLY` parameter is set to 2.

The Verilator output is:
```
$ make TOP=tcb_vip_tb
verilator --binary -O3 -j 0 --timescale 1ps/1ps --x-assign fast -Wno-INITIALDLY --assert --trace-fst --trace-structs   --top tcb_vip_tb ../../hdl/rtl/tcb_if.sv ../../hdl/rtl/lib/tcb_lib_pas.sv ../../hdl/rtl/lib/tcb_lib_reg.sv ../../hdl/rtl/lib/tcb_lib_arb.sv ../../hdl/rtl/lib/tcb_lib_mux.sv ../../hdl/rtl/lib/tcb_lib_dec.sv ../../hdl/rtl/lib/tcb_lib_dmx.sv ../../hdl/rtl/lib/tcb_lib_err.sv ../../hdl/rtl/gpio/tcb_gpio.sv ../../hdl/rtl/uart/tcb_uart_ser.sv ../../hdl/rtl/uart/tcb_uart_des.sv ../../hdl/rtl/uart/tcb_uart_fifo.sv ../../hdl/rtl/uart/tcb_uart.sv ../../hdl/tbn/vip/tcb_vip_pkg.sv ../../hdl/tbn/vip/tcb_vip_mon.sv ../../hdl/tbn/vip/tcb_vip_sub.sv ../../hdl/tbn/vip/tcb_vip_man.sv ../../hdl/tbn/vip/tcb_vip_tb.sv ../../hdl/tbn/lib/tcb_lib_pas_tb.sv ../../hdl/tbn/lib/tcb_lib_reg_tb.sv ../../hdl/tbn/lib/tcb_lib_mux_tb.sv ../../hdl/tbn/lib/tcb_lib_dmx_tb.sv ../../hdl/tbn/gpio/tcb_gpio_tb.sv ../../hdl/tbn/uart/uart_model.sv ../../hdl/tbn/uart/uart_model_tb.sv ../../hdl/tbn/uart/tcb_uart_tb.sv
make[1]: Entering directory '/home/izi/VLSI/rp32/submodules/tcb/sim/verilator/obj_dir'
ccache g++  -I.  -MMD -I/home/izi/VLSI/rp32/submodules/tcb/../verilator/include -I/home/izi/VLSI/rp32/submodules/tcb/../verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_TIME_CONTEXT  -fcoroutines -std=gnu++17 -Os -c -o verilated.o /home/izi/VLSI/rp32/submodules/tcb/../verilator/include/verilated.cpp
ccache g++  -I.  -MMD -I/home/izi/VLSI/rp32/submodules/tcb/../verilator/include -I/home/izi/VLSI/rp32/submodules/tcb/../verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_TIME_CONTEXT  -fcoroutines -std=gnu++17 -Os -c -o verilated_fst_c.o /home/izi/VLSI/rp32/submodules/tcb/../verilator/include/verilated_fst_c.cpp
ccache g++  -I.  -MMD -I/home/izi/VLSI/rp32/submodules/tcb/../verilator/include -I/home/izi/VLSI/rp32/submodules/tcb/../verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_TIME_CONTEXT  -fcoroutines -std=gnu++17 -Os -c -o verilated_timing.o /home/izi/VLSI/rp32/submodules/tcb/../verilator/include/verilated_timing.cpp
/usr/bin/perl /home/izi/VLSI/rp32/submodules/tcb/../verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vtcb_vip_tb.cpp Vtcb_vip_tb___024root__DepSet_h0d12d57a__0.cpp Vtcb_vip_tb___024root__DepSet_h8fdf1f90__0.cpp Vtcb_vip_tb_tcb_if__A20_D20_DB2__DepSet_ha21af8f9__0.cpp Vtcb_vip_tb_tcb_if__A20_D20_DB2__DepSet_h30e73c0f__0.cpp Vtcb_vip_tb__main.cpp Vtcb_vip_tb__Trace__0.cpp Vtcb_vip_tb___024root__Slow.cpp Vtcb_vip_tb___024root__DepSet_h0d12d57a__0__Slow.cpp Vtcb_vip_tb___024root__DepSet_h8fdf1f90__0__Slow.cpp Vtcb_vip_tb_tcb_if__A20_D20_DB2__Slow.cpp Vtcb_vip_tb_tcb_if__A20_D20_DB2__DepSet_h30e73c0f__0__Slow.cpp Vtcb_vip_tb__Syms.cpp Vtcb_vip_tb__Trace__0__Slow.cpp > Vtcb_vip_tb__ALL.cpp
echo "" > Vtcb_vip_tb__ALL.verilator_deplist.tmp
ccache g++  -I.  -MMD -I/home/izi/VLSI/rp32/submodules/tcb/../verilator/include -I/home/izi/VLSI/rp32/submodules/tcb/../verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_TIME_CONTEXT  -fcoroutines -std=gnu++17 -Os -c -o Vtcb_vip_tb__ALL.o Vtcb_vip_tb__ALL.cpp
Archive ar -rcs Vtcb_vip_tb__ALL.a Vtcb_vip_tb__ALL.o
g++    verilated.o verilated_fst_c.o verilated_timing.o Vtcb_vip_tb__ALL.a   -lz    -o Vtcb_vip_tb
rm Vtcb_vip_tb__ALL.verilator_deplist.tmp
make[1]: Leaving directory '/home/izi/VLSI/rp32/submodules/tcb/sim/verilator/obj_dir'
obj_dir/Vtcb_vip_tb 
else: DLY =          2
@TOP.tcb_vip_tb.mon: TCB log: W: adr=0x00000000 ben=0b1111 dat=0x01234567 txt="#Eg" err=0
@TOP.tcb_vip_tb.mon: TCB log: R: adr=0x00000000 ben=0b1111 dat=0x00000000 txt="    " err=0
- ../../hdl/tbn/vip/tcb_vip_tb.sv:80: Verilog $finish
```

As reference I used the Vivado simulator, which created the next output:
```
Time resolution is 1 ps
else: DLY =          2
for: DLY =          2, genvar i =           1
DLY =          2, genvar i =           1
DLY =          2, genvar i =           1
DLY =          2, genvar i =           1
DLY =          2, genvar i =           1
DLY =          2, genvar i =           1
DLY =          2, genvar i =           1
@tcb_vip_tb.mon: TCB log: W: adr=0x00000000 ben=0b1111 dat=0x01234567 txt="#Eg" err=0
DLY =          2, genvar i =           1
DLY =          2, genvar i =           1
DLY =          2, genvar i =           1
@tcb_vip_tb.mon: TCB log: R: adr=0x00000000 ben=0b1111 dat=0x76543210 txt="vT2" err=0
DLY =          2, genvar i =           1
DLY =          2, genvar i =           1
DLY =          2, genvar i =           1
DLY =          2, genvar i =           1
$finish called at time : 260 ns : File "/home/izi/VLSI/rp32/submodules/tcb/hdl/tbn/vip/tcb_vip_tb.sv" Line 80
```

wsnyder, 2022-11-03:
Thanks for the verilator format test.  That test passes on the current master, and I don't see anything obviously wrong, is it supposed to fail?

jeras, 2022-11-03:
The test is supposed to pass, but my code which does something similar fails. This means the problem is a bit more complex to reproduce. I will try to remove elements from my code till I am able to create a minimal failing test, but it will take me some time.

Since this is something rather important for my very generic VIP and library code, I will probably invest my time to further debug this issue.

jeras, 2022-11-06:
I was able to create a test reproducing the issue, and it has some interesting output:
```
%Error: /home/izi/VLSI/verilator/test_regress/../include/verilated_timing.cpp:60: %Error: Encountered process that should've been resumed at an earlier simulation time. Missed a time slot?
```

The commit with the test is here:
https://github.com/verilator/verilator/commit/8d94314e658c0cf214f5c8369863e2cae62b0f32

The name of the testcase is based on my first attempt to identify the issue, but it is probably unrelated.
I could probably reduce the code further, but again, I am not sure what to focus on.

wsnyder, 2022-11-10:
If you could please reduce this further, I'm not sure what I'm even looking for in that file. Thanks.

jeras, 2022-11-10:
OK, I will reduce it further.

wsnyder, 2023-01-26:
Note this is still waiting on a small test case, closing until received.

closed 2023-01-26 by wsnyder