verilator/verilator #3726: Internal Error: ../V3DfgPeephole.cpp:1387: Width mismatch

shtaxxx, 2022-11-05:
I faced an internal error on Verilator 5.002 on macOS Monterey 12.6.1. (Verilator was installed via homebrew).
I have tested the master branch version (8d61cea366057f0a8537d09168a3377f095784ab) of Verilator, but I got the same result.

I was compiling a generated Verilog HDL source code by my DNN hardware synthesis compiler https://github.com/NNgen/nngen (v1.3.4-rc branch, c2e49419412558b0c16df678fa6d331e8af63f6c).

This is a minimal example obtained from the original problem code. (`test.v`)

``` verilog
module top
  (
   input [7:0] din,
   output [7:0] dout
   );

  wire cond;
  assign cond = din > 1 && din < 254;

  wire tmp_out;
  assign tmp_out = ((cond)? 0 : 1) && 1;

  assign dout = tmp_out? 100 : 200;

endmodule
```

I executed the following command.

```bash
verilator --lint-only test.v
```

The following result was obtained.

```
%Error: Internal Error: test.v:12:27: ../V3DfgPeephole.cpp:1387: Width mismatch
   12 |   assign tmp_out = ((cond)? 0 : 1) && 1;
      |                           ^
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

Thank you very much for the development of the wonderful hardware tool.

shtaxxx, 2022-11-05:
If I modify the description a little bit, it works well.
``` verilog
  //assign tmp_out = ((cond)? 0 : 1) && 1; // NG
  //assign tmp_out = ((cond)? 1'b0 : 1) && 1; // NG
  assign tmp_out = ((cond)? 2'b0 : 1) && 1; // OK
```

gezalore, 2022-11-05:
Another instance of an earlier optimization creating something not quite well formed and the new DFG complaining about it. Now fixed on master.

Thank you for the neat bug report.

closed 2022-11-05 by gezalore

shtaxxx, 2022-11-05:
Thank you very much for the quick update!

commit 6ae6b162 (2022-11-05): V3Const: Fix folding of LogAnd with non-bool operands

Folding an AstLogAnd with a non-zero constant operand used to coerce the
type of the other operand, yielding an ill-typed node that DFG was then
unhappy about. Add a RedOr instead if the width of the replacement
operand is greater than zero.

Fixes #3726