verilator/verilator #3728: Some AsciiDoc code is not translated correctly due to line breaks

jeras, 2022-11-06:
One example is:
```asciidoc
   Specifies optional object or library files to be linked in with the
   Verilog code, as a shorthand for :vlopt:`-LDFLAGS \<file\>
   <-LDFLAGS>`. The file path should either be absolute, or relative to
...
```
In the generated file only the part of `:vlopt:` in the first line is present in the [generated documentation page](https://veripool.org/guide/latest/exe_verilator.html?highlight=specifies%20optional%20object%20library%20files).

To me it seems the issue was caused by using a tool which breaks source code lines to a length of 80 characters.
If this issues are fixed, rerunning the tools would probably bring some of them back.

I started a branch where I started with fixing a code example, than got carried away with cosmetic changes and so I noticed this issue.
I wish to create a pull request for this branch but I am unsure if it would be accepted, since:
- there are a lot of changes, 99% of them of cosmetic nature (using fixed width fonts, ...),
- nether the original docs or my changes are strict about which to use where `:code:`, `:vlopts:`, `:command:`, `:file:`, ...,
- I probably introduced some errors of my own.

I would like to know, if this kind of documentation changes would be accepted, so I can decide whether to continue or not.
Unfortunately there is just too much text to read through for me to be able to split this changes into multiple easy to review commits.

Please have a look at the mentioned branch:
https://github.com/jeras/verilator/tree/docs-3725

wsnyder, 2022-11-06:
I think most all your cleanups are good improvements.  The only item I might not do is use :markups in the headings, as I suspect that will cause problems rendoring (but you can check).

As long as you're no worse with using vlopt than the original that is fine.  Make sure there's no warnings printed when you do a "make info" please.

I'd suggest make a pull request for everything but the line wrapping.  For that, we should see if there's some better tool setting to fix the issue, or perhaps write a bit of python to de-wrap the lines ourselves.  I'd prefer not to keep long lines in the sources themselves, unless there's an emacs auto-filler than can do the right thing and keep them.

jeras, 2022-11-06:
OK, I already got the documentation build to work,
so I will be able to check a few representative changes,
and skim the rest.
And I will break the work into multiple pull requests,
each targeting a few files,
to avoid too many marge conflicts.

I personally find the 80 character line limit obsolete,
and usually target something closer to 80~150 characters for documentation
in Markdown AsciiDoc or ReStructuredText format.
I also try to place the newline so it breaks the text into sensible units,
I did not study advanced English grammar at school (my 3rd language),
so I am not sure how to call the units, they are related to punctuation.
I find this approach reduces the need to redo line breaks when the text is updated.

In executable source code (SW/HDL) I often write longer lines,
where I see an advantage in using block (vertical) edit mode on the lines.

wsnyder, 2022-12-11:
Think the issue here is now resolved, and tested for, plus I made some grammar cleanups, so I'm closing.  If you have other text cleanups they would be appreciated, please make a pull request based off the most recent master, thanks.

closed 2022-12-11 by wsnyder