verilator/verilator #3730: Timing support / Transport delay versus Inertial delay

mathieu-tp, 2022-11-07:
Delayed assignments such as 
**assign  #12ns outsig = insig;** 

are handled by commercial tools as **inertial delay**:  I.e glitches on the **insig** signal are filtered out if there duration is less than 12ns. It's correct for models of logic gates, it should not be used for delay line models.

Verilator seems to handle such assignments as **transport delay** : *outsig* is just a clone of *insig* with a 12ns delay (delay line model).

Is it a known problem ?  Is it even a problem ? 

Perhaps documentation should explain such difference of behavior.

mtdudek, 2022-11-07:
I believe this issue was risen in issue #3711.

mathieu-tp, 2022-11-07:
OK, it's the same issue

closed 2022-11-07 by mathieu-tp