verilator/verilator #3732: Question: SDF annotation

Blebowski, 2022-11-08:
Hello, 

I would like to ask whether there is an intent to add SDF annotation. Since now verilator handles simulation of
delays, this might be very interesting for pre-TO Sign-off simulation.

Despite strong shift towards sign-off with STA only, we still go GLS before TO (and will continue to do so :), good
reasons can be found e.g. at: http://www.deepchip.com/items/0569-01.html)

wsnyder, 2022-11-08:
Nope, but was worth asking ;)

Short term, the delay model is not designed to handle millions of delays and will not perform acceptably.  Longer term I don't want SDF code as the low use does not offset the maintenance cost.  So this is a use case well left to the closed source tools.

FWIW SDF is not realistic on large designs because it is too slow to be able to run enough stimulus to find problems.  I haven't been involved in any SDF annotated chips since 2005, and have had dozens of multi-billion gate tape outs since then. They were bug free because we have appropriate STA, CDC, non-SDF gate sims, formal equivalence checking, reset formal tools, and multicycle and false path modules and checkers necessary to make sure there are no bug escapes.  These at least double cover every item mentioned in the deepchip article without SDF.

closed 2022-11-08 by wsnyder

Blebowski, 2022-11-08:
Hello @wsnyder, thanks for quick reply and explanation. So, even if I started to dive into verilator code-base and started working on SDF annotator, you would not accept it in mainline / master?

I understand that SDF gate sims might be slow on large chips (I work on designs with less than 10M gates),
but for smaller companies, the cost of extra tools (CDC/RDC/Formal) might be a no-go. Also for designs which
are not "large chunks of synchronous logic", but use asynchronous design techniques, SDF gate sims are
irreplacable.

wsnyder, 2022-11-08:
I agree there are some usages for SDF, just as there are some critical uses for Verilog-AMS and for UPF power methodologies, but at this point Verilator's picking its investments.

Hypothetically if a SDF patch set was reasonably small and maintainable, had a good test suite, and there were suitable performance enhancements (I think these are general and not just SDF) so it performed at a closed-source competitive level, I'd take it.  This is a good sized effort, I'd SWAG several person-months.

Blebowski, 2022-11-09:
OK, thanks, I am intrigued by the idea to start working on it, but I think its at least year of work, so I will see.