verilator/verilator #3737: --lint-only Unsupported: Unpacked array in packed struct

jjones-github, 2022-11-09:
This is a relatively small example and this warning can be disabled. I think I am more trying to understand if this is identifying a potential real issue or maybe being very pessimistic and saying "hey, some tools might do something you don't expect if accessing via bit location". We (small team of 2) do plan to only access members of structures by member not by bit location so we do believe this will not be an issue in our design.

The question would be is this something worth improving or is it something we should just live with and understand? I tried to run this on edaplayground also but I have not been approved for access to check it out. We plan to try it in Vitis in the near future too just as another check.

`exp.sv`
```verilog
module exp
// Import package statements
import expPackage::*;
(
    input clk, rstN
);

endmodule // exp
```
`expPackage.sv`
```verilog
package expPackage;
`include "exp.svh"
endpackage //expPackage
```
`exp.svh`
```verilog
localparam ASIZE =                            1;  // The size of A
localparam ASIZE2 =                           2;  // The size of A

// types
typedef logic[2-1:0] twoBitT; //this is a 2 bit type
typedef logic[3-1:0] threeBitT; //this is a 3 bit type
typedef logic[4-1:0] fourBitT; //this is a 4 bit type
typedef logic[ASIZE-1:0] aSizeT; //type of width ASIZE
typedef logic[ASIZE2-1:0] aBiggerT; //yet another type

// structures
typedef struct packed {
    aSizeT variablea[ASIZE2]; //One bit of A
    twoBitT variablea2; //
} aSt ;

typedef struct packed {
    aSizeT variablea; //One bit of A
} aASt ;

typedef struct packed {
    twoBitT variablec; //Two bits of C
    threeBitT variablec2; //Three bits of C
} seeSt ;

typedef struct packed {
    threeBitT variabled; //Three bits of D
    fourBitT variabled2; //Four bits of D
} dSt ;

typedef struct packed {
    aSizeT variablea; //One bit of A
    dSt bob; //
    seeSt joe[2]; //
} nestedSt ;
```
Command:
`verilator --lint-only -sv expPackage.sv exp.sv --top-module exp -y .`
Output:
```%Warning-UNPACKED: exp.svh:13:21: Unsupported: Unpacked array in packed struct/union (struct/union converted to unpacked)
   13 |     aSizeT variablea[ASIZE2];  
      |                     ^
                   expPackage.sv:2:19: ... note: In file included from expPackage.sv
                   ... For warning description see https://verilator.org/warn/UNPACKED?v=5.002
                   ... Use "/* verilator lint_off UNPACKED */" and lint_on around source to disable this message.
%Warning-UNPACKED: exp.svh:34:14: Unsupported: Unpacked array in packed struct/union (struct/union converted to unpacked)
   34 |     seeSt joe[2]; 
      |              ^
                   expPackage.sv:2:19: ... note: In file included from expPackage.sv
%Error: Exiting due to 2 warning(s)
        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```
Adding the `-Wno-UNPACKED` of course removes the warnings as expected. The second struct certainly seems packed to me since the `seeSt` that `joe` is an array of 2 of is packed itself. The other seems packable too. Unless there is something in the LRM that disqualifies these as being packed, or packable. 

If this has potential for improvement, great. If this is the way it is we will work around it since we think our usage does not add risk of issues down the road. If there is any clarity that maybe isn't obvious that is also appreciated. Thank you.

udif, 2022-11-09:
IEEE1800-2017 page 140:
> Only packed data types and the integer data types summarized in Table 6-8 (see 6.11) shall be legal in
packed structures. 

It's important to understand the concepts behind packed vs unpacked items:
* packed items (structs or arrays) are just syntactic sugar for vectors. Any bit combination in any packed struct/array can change at any time.
* For unpacked arrays you may only change a single entry at a time, and they map nicely to RAM/ROM memory.

packed structures are just a nice way to collect related wires/flops, but physically each one is individually addressable. It is obvious you can't guarantee this by having an unpacked array inside a packed structure.

jjones-github, 2022-11-09:
I think the main issue was not noticing that this style `someType someVariable[2]` inferred as unpacked where `someType [1:0] someVariable` is packed. Language interpretation problem on my part. So to alter the example above.

This modification is lint clean.
```verilog
localparam ASIZE =                            1;  // The size of A
localparam ASIZE2 =                           2;  // The size of A

// types
typedef logic[2-1:0] twoBitT; //this is a 2 bit type
typedef logic[3-1:0] threeBitT; //this is a 3 bit type
typedef logic[4-1:0] fourBitT; //this is a 4 bit type
typedef logic[ASIZE-1:0] aSizeT; //type of width ASIZE
typedef logic[ASIZE2-1:0] aBiggerT; //yet another type

// structures
typedef struct packed {
    aSizeT [ASIZE2-1:0] variablea; //One bit of A
    twoBitT variablea2; //
} aSt ;

typedef struct packed {
    aSizeT variablea; //One bit of A
} aASt ;

typedef struct packed {
    twoBitT variablec; //Two bits of C
    threeBitT variablec2; //Three bits of C
} seeSt ;

typedef struct packed {
    threeBitT variabled; //Three bits of D
    fourBitT variabled2; //Four bits of D
} dSt ;

typedef struct packed {
    aSizeT variablea; //One bit of A
    dSt bob; //
    seeSt [1:0] joe; //
} nestedSt ;
```

I was also expecting this to be lint clean too but it is not. Dose this code look like it should lint or should it still be UNPACKED? The structure typedef(s) are not declared as packed.
```verilog
localparam ASIZE =                            1;  // The size of A
localparam ASIZE2 =                           2;  // The size of A

// types
typedef logic[2-1:0] twoBitT; //this is a 2 bit type
typedef logic[3-1:0] threeBitT; //this is a 3 bit type
typedef logic[4-1:0] fourBitT; //this is a 4 bit type
typedef logic[ASIZE-1:0] aSizeT; //type of width ASIZE
typedef logic[ASIZE2-1:0] aBiggerT; //yet another type

// structures
typedef struct {
    aSizeT variablea[ASIZE2]; //One bit of A
    twoBitT variablea2; //
} aSt ;

typedef struct {
    aSizeT variablea; //One bit of A
} aASt ;

typedef struct {
    twoBitT variablec; //Two bits of C
    threeBitT variablec2; //Three bits of C
} seeSt ;

typedef struct {
    threeBitT variabled; //Three bits of D
    fourBitT variabled2; //Four bits of D
} dSt ;

typedef struct {
    aSizeT variablea; //One bit of A
    dSt bob; //
    seeSt joe[2]; //
} nestedSt ;
```

wsnyder, 2022-11-10:
This code I believe is legal System Verilog, but now you are truly hitting the unsupported unpacked issue.  You can suppress that warning to make progress.

I'm closing as think we answered your questions, but feel free to followup if needed.

closed 2022-11-10 by wsnyder

wsnyder, 2022-11-10:
It is throwing a warning because this is not supported by verilator, please see the URL it mentions https://verilator.org/warn/UNPACKED

jjones-github, 2022-11-16:
> It is throwing a warning because this is not supported by verilator, please see the URL it mentions https://verilator.org/warn/UNPACKED

I did read that a few days ago but missed the fact that unpacked is not supported. I was still trying to figure out my issue. I assume this isn't something on the list to support? I can imagine it might be a large refactor with DPI, VPI, and the built in api.

wsnyder, 2022-11-16:
Supporting unpacked is #1541, it's not a large refactor but like most things just need more authors to get things done (hint).

jjones-github, 2022-11-17:
> Supporting unpacked is #1541, it's not a large refactor but like most things just need more authors to get things done (hint).

;), if I run into spare time I will attempt to help. Just founded a startup several moths ago so that's been pull down a lot of hours.