verilator/verilator #3740: Internal fault when constructing sequence of assignment

rachitnigam, 2022-11-10:
The following program produces an internal fault when running `verilator --lint-only`:
```
`default_nettype none

module add_one ();
    logic out_write_en;
    logic reg_to_mem_go_in;
    logic reg_to_mem_done_out;
    initial begin
        out_write_en = 1'd0;
        reg_to_mem_go_in = 1'd0;
    end
    assign out_write_en = reg_to_mem_go_in;
    assign reg_to_mem_go_in = ~reg_to_mem_done_out;
endmodule
```
Output:
```
% verilator --lint-only out.sv        
%Error: Verilator internal fault, sorry. Suggest trying --debug --gdbbt
%Error: Command Failed /usr/local/Cellar/verilator/5.002/bin/verilator_bin --lint-only out.sv
```

Version: Verilator 5.002 2022-10-29 rev UNKNOWN.REV
What OS and distribution are you using? MacOS Ventura

wsnyder, 2022-11-10:
@gezalore, this dies in Dfg, can you look please?  It works in 4.228.

gezalore, 2022-11-12:
Fix is in master. Which also hopefully covers all other non-Dfg / Dfg multi-driven issues. Will file a PR for Dfg/Dfg multi-driver resolution separately.

commit dbcaad99 (2022-11-12): Dfg: Fix crash on additional driver from non-DFG logic

Ensure variables written by non-DFG code are kept

Fixes #3740
closed 2022-11-12 by gezalore