verilator/verilator #3742: build verilator using msvc

codelec, 2022-11-12:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
https://github.com/codelec/verilator/commit/215abaf242836d10b05cd6aa9701f1a6a5c8f78d
https://github.com/codelec/verilator/commit/9f4e1b38d4622594a3e665dc134e038ac84bbe92 
Inorder to successfully build verilator using msvc on windows i need to make the above changes
What 'verilator --version' are you using?  Did you try it with the git master version?
I tried with git master 
What OS and distribution are you using?
Windows 11
May we assist you in trying to fix this in Verilator yourself?
I can make a PR if the changes are needed

codelec, 2022-11-12:
I am using cmake as the build tool and msvc as the compiler

wsnyder, 2022-11-12:
Thanks for debugging, makes sense. Would you mind making this into a pull?  Thanks.

codelec, 2022-11-12:
I will make a PR

wsnyder, 2022-11-13:
Thanks for your PR, merged with appreciation.

closed 2022-11-13 by wsnyder