verilator/verilator #3745: CASEINCOMPLETE warning on unique case with enum

gkamendje, 2022-11-12:
Using the following code example (taken from https://www.edaplayground.com/x/GQ2E) 
```
module fsm(input logic X, 
           input logic reset,
           input logic clk,
           output logic Z
          );

  enum logic [2:0] {S0,S1,S2,S3,S4,S5} State, Next;
  
  always @(posedge clk or posedge reset)
    begin
      if ( reset==1'b1 )
        begin
          State <= S0;
        end
      else 
        begin
         State <= Next;
        end
    end

  always_comb begin: set_next_state
    
    Next = State; //default value
    unique case ( State )
      S0: if (X==1'b1) 
           Next = S1;
          else
            Next = S0;
      
      S1: if (X==1'b0) 
              Next = S2;
          else if (X==1'b1)
              Next = S1;
      
      S2: if (X==1'b1) 
              Next = S3;
           else if (X==1'b0)
              Next = S0;
      
      S3: if (X==1'b1) 
              Next = S4;
           else if (X==1'b0)
              Next = S2;
      
      S4: if (X==1'b0) 
              Next = S5;
          else if (X==1'b1)
              Next = S1;
      
      S5: if (X==1'b0) 
              Next = S0;
          else if (X==1'b1)
              Next = S3;
      	  
    endcase
  
  end: set_next_state
  
  always_comb begin: set_output
    
    unique case ( State ) 
      S0,S1,S2,S3,S4: Z = 0;
      S5: 
        Z = 1;
      default:
        Z = 1'bX;
    endcase
  
    
  end: set_output
  
endmodule

```
and running 

`verilator --lint-only  fsm.sv  --top-module fsm`
verilator issues the following warning message
```
%Warning-CASEINCOMPLETE: design.sv:26:12: Case values incompletely covered (example pattern 0x6)
   26 |     unique case ( State )
      |            ^~~~
                         ... For warning description see https://verilator.org/warn/CASEINCOMPLETE?v=5.003
                         ... Use "/* verilator lint_off CASEINCOMPLETE */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)

```
which in my option is not correct given that the `unique case` statement is used in conjunction with an **enum** which limits the possible values of the variable `State`. As a matter of fact, the example pattern **0x6** reported in the warning message is not a legal pattern as dictated by the **enum**.

What 'verilator --version' are you using?  

`Verilator 5.003 devel rev v5.002-29-g0a045a7bf`

Did you try it with the git master version? 
yes

Which Os are you using?

`Centos7`

May we assist you in trying to fix this in Verilator yourself?
Yes

wsnyder, 2022-11-12:
I can see why that might seem logical. However the warning is intentional; in real hardware the flops containing State might be one of the undefined values, and if your design powers up in such a state where state vector is not one of the known states you will have a potential problem.  It's advisable to always have an e.g. "default : $error("bad case");" or whatever your methodology preference is in cases such as these.

gkamendje, 2022-11-12:
My understanding is that the  "default : $error("bad case");" was required in Verilog. However, `section 12.5.3` of the `1800-2017 SystemVerilog LRM` has a note on page 307 which reads as follows: 
```
NOTE—By specifying unique or priority, it is not necessary to code a default case to trap unexpected case
values. 
```
I wonder how this should be interpreted in this context.

wsnyder, 2022-11-12:
Good point. I believe that is said because a tool is supposed to enter a runtime warning if the unique is violated.  Maybe that's a reasonable action here, that is suppress the compile time warning and leave it for a runtime error.  Let me think about that.

wsnyder, 2022-11-13:
There's already code for the runtime assertion, so I think the only item is to suppress the warning.

Would you be willing to try to make a pull request? - Make a test case, then fix V3Case to make sure it gets suppressed, also make another test where the case enum unique is forced to a value outside the enum and make sure the assertion fires.

gkamendje, 2022-11-13:
I was considering commenting out lines `232` to `234` in `VCase.cpp` as suggested. However, I was wondering if the warning should not be firing when the **unique** keyword is not used in the case statement. Is this case handle somewhere else?

wsnyder, 2022-11-13:
I think the fix is when the case data type is an enum, if all the provided values match all the values within the enum, then suppress.  We don't want to disable the warning on non-enum uniques.

gkamendje, 2022-11-13:
I see how to access some case information using `nodep->uniquePragma()` and similar functions. How do I access the case data type?

wsnyder, 2022-11-13:
Something like...  This will only execute if it the case's expression is an enum.

   if (AstENumDType* enumDtp = VN_CAST(casep->exprp()->dtypeSkipRefp(), EnumDType)) { .... }

gkamendje, 2022-11-13:
With `bool isCaseTreeFast(AstCase* nodep)` and `if (AstEnumDType* enumDtp = VN_CAST(nodep->exprp()->dtypeSkipRefp(), EnumDType)) {...}` and I am getting 

```
in file included from ../V3Case.cpp:42:
../V3Case.cpp: In member function ‘bool CaseVisitor::isCaseTreeFast(AstCase*)’:
../V3Case.cpp:201:73: error: ‘class AstNode’ has no member named ‘dtypeSkipRefp’
  201 |                     if (AstEnumDType* enumDtp = VN_CAST(nodep->exprp()->dtypeSkipRefp(), EnumDType)) {
      |                                                                         ^~~~~~~~~~~~~

```
Not sure what I am missing here.

wsnyder, 2022-11-13:
Should be exprp()->dtypep()->skipRefp()

gkamendje, 2022-11-13:
Any hint on how to access the enum values?

gkamendje, 2022-11-13:
I think I still need some guidance for the check condition. The following in `V3Case.cpp` does not seem to work.
```  
     if (AstEnumDType* enumDtp = VN_CAST(nodep->exprp()->dtypep()->skipRefp(), EnumDType)) {
                    cout << "Case is enum, check if all values are within the enum"  << endl;
        } else {
            for (uint32_t i = 0; i < numCases; ++i) {
                if (!m_valueItem[i]) {
                    nodep->v3warn(CASEINCOMPLETE, "Case values incompletely covered "
                                                  "(example pattern 0x"
                                                      << std::hex << i << ")");
                    m_caseNoOverlapsAllCovered = false;
                    return false;
                }
            }
        }
```

I am using the  following testcase. 

```
module t (
    input logic in 
  );

  enum logic [1:0] {S0,S1,S2} state, next;

  always_comb begin: set_next_state
    next = state; //default value
    unique case ( state )
      S0: if (in==1'b1) 
           next = S1;
          else
           next = S0;
      S1: if (in==1'b0) 
            next = S2;
          else if (in==1'b1)
            next = S1;
      S2: if (in==1'b1) 
            next = S1;
          else if (in==1'b0)
            next = S2;
    endcase
  
  end: set_next_state

endmodule
```

wsnyder, 2022-11-13:
I'm not sure, look a the .tree files, and dump the pointer to see what it is pointing to if not an enum.
Be sure to have a test which uses enums but not all of them, so the warning comes up.

To see enum values 

     enump->itemsp()->foreach([](const AstEnumItem* itemp) { ... }

Note this is a lambda so you need to pass appropriate information in. If that's confusing use a loop like in V3EmitCImp.cpp:emitTraceDeclDType.

gkamendje, 2022-11-14:
I am still having hard time detecting when the case is an enum. I would need some more guidance on how to proceed. 
Could you please have a look at the following commit? 
https://github.com/verilator/verilator/compare/master...gkamendje:verilator:enum_caseincomplete%233745
NB:
`test_regress/t/t_case_enum_incomplete_bad.*` are just place holder for now.

wsnyder, 2022-11-17:
It should be skipRefToEnump() with that it matches:

Case is an enum, TODO: check if all values are within the enum
"S0"
"S1"
"S2"

gkamendje, 2022-11-22:
I was trying to use the following code to iterate through the case items and hence compare them with entries in the enum list. 
```
 if (AstEnumDType* enumDtp = VN_CAST(nodep->exprp()->dtypep()->skipRefToEnump(), EnumDType)) {
            cout << "Case is an enum. Checking if all values are within the enum"  << endl;
            for (uint32_t i = 0; i < numCases; ++i) {
                for (AstEnumItem* itemp = enumDtp->itemsp(); itemp; itemp = VN_AS(itemp->nextp(), EnumItem)) {
                      AstConst* const constp = VN_AS(itemp->valuep(), Const);
                      const AstNode* const itemv = m_valueItem[i];
                         cout << "Current value " << itemv << " => " <<  itemp->prettyName() << " numeric value "<< constp->num().toDecimalU() << endl;
                }
            }
        }
```
However, i am not getting the expected numeric value of the case item and the last entries seems to be null pointers. Any hint on accessing the numeric value? 
```
Case is an enum. Checking if all values are within the enum
Current value CASEITEM 0x2efab40 {d16aj} => S0 numeric value 0
Current value CASEITEM 0x2efab40 {d16aj} => S1 numeric value 1
Current value CASEITEM 0x2efab40 {d16aj} => S2 numeric value 2
Current value CASEITEM 0x2efb240 {d20aj} => S0 numeric value 0
Current value CASEITEM 0x2efb240 {d20aj} => S1 numeric value 1
Current value CASEITEM 0x2efb240 {d20aj} => S2 numeric value 2
Current value CASEITEM 0x2f74ab0 {d24aj} => S0 numeric value 0
Current value CASEITEM 0x2f74ab0 {d24aj} => S1 numeric value 1
Current value CASEITEM 0x2f74ab0 {d24aj} => S2 numeric value 2
Current value nullptr => S0 numeric value 0
Current value nullptr => S1 numeric value 1
Current value nullptr => S2 numeric value 2
```

gkamendje, 2022-11-26:
I have created PR #3782 that suppresses the CASEINCOMPLETE warning when the case is a unique/priority enum.  However, forcing the unique case enum to a value outside the enum does not trigger a runtime assertion. Please have a look at this since it is beyond what I feel I could do.

wsnyder, 2022-12-01:
Merged, thanks for your work on this.

closed 2022-12-01 by wsnyder