verilator/verilator #3747: internals documentation not generated properly

gkamendje73, 2022-11-14:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
The following link that points to the documentation of the `internals` is broken. 
https://veripool.org/guide/latest/internals.rst

The `internals` documentation is not generated either when running `make docs`

What 'verilator --version' are you using?  Did you try it with the git master version?
Latest (true on the latest website as well)
What OS and distribution are you using?
Centos 7
May we assist you in trying to fix this in Verilator yourself?
Yes

wsnyder, 2022-11-17:
The assumption, perhaps incorrect, was people would be unlikely to look at a PDF of internals since it is intended for developers that can read the source, or alternatively look at https://github.com/verilator/verilator/blob/master/docs/internals.rst

The veripool link should be fixed to redirect to github.

wsnyder, 2022-11-19:
Assuming above answers, if you think we need more feel free to comment/reopen, thanks.

closed 2022-11-19 by wsnyder

gkamendje73, 2022-11-21:
Thanks for the fix. Reading the internal docu in an web browser is way easier.