verilator/verilator #3749: Syntax error in generated code if the return value of `array.first()` method is not read & related wrong simulation behavior

goekce, 2022-11-15:
```systemverilog
module m;

logic clk = 0;
logic mem [int];
int i;

initial begin
	mem[42] = 1;
	mem.first(i);
	$display("initial end. i: %d", i);
end

always #1 clk = ~clk;

always @(posedge clk)
	begin
	mem.first(i);
	// leads to compiler error

	i = mem.first(i);
	// ok

	$finish;
	end
endmodule
```
from LRM 7.9.4:
```
The first() method assigns to the given index variable the value of the first (smallest) index in the associative array. It returns 0 if the array is empty; otherwise, it returns 1.
```

- 5.002 2022-10-29, did not try with master
- Archlinux
- Fix assistance welcome

---

This is actually part of a bigger problem that I encountered where merely including `mem.first()` (not even executed) leads to wrong simulation results. If I include `mem.first()`, then the combinational path is evaluated twice, otherwise only once. Here is an excerpt from my debugging session.

If I include `i = mem.first(cycle);` in my testbench:
```
cycle           1:
cycle           2:
cycle           3:
mi: ead08093, OP: (rf[ 1] = 00000000, rfn = 00000000) + 4294966957
mi: ead08093, OP: (rf[ 1] = fffffead, rfn = fffffead) + 4294966957
cycle           4:
cycle           5:
mi: f5610113, OP: (rf[ 2] = ff76e000, rfn = ff76e000) + 4294967126
mi: f5610113, OP: (rf[ 2] = ff76df56, rfn = ff76df56) + 4294967126
...
```

`rf[]` should be updated using `rfn[] + shown_integer` in the next cycle, but the update happens in the same cycle before the values have settled (`ffff_fead` = `4294966957`).

If I comment `i = mem.first(cycle);` in my testbench:

```
cycle           1:
cycle           2:
cycle           3:
mi: ead08093, OP: (rf[ 1] = feedc000, rfn = feedc000) + 4294966957
cycle           4:
cycle           5:
mi: f5610113, OP: (rf[ 2] = ff76e000, rfn = ff76e000) + 4294967126
...
```
then the behavior is correct.

I have the feeling that an additional evaluation step is introduced if `first()` method is used but the registers get updated too early. What do you think?

I can try to isolate this second problem if needed.

gezalore, 2022-11-15:
This was recently changed on master and likely works, could you please try?

goekce, 2022-11-15:
Thanks Geza, master fixes the issue.

closed 2022-11-15 by goekce