verilator/verilator #3750: Unexpected simulation behavior if `array.first()` method is included in code (not used though)

goekce, 2022-11-15:
Merely including `mem.first()` (not executed though) leads to wrong simulation results. If I include `mem.first()`, then the combinational path is evaluated twice, otherwise only once. Here is an excerpt from my debugging session where I display relevant values at the end of a combinational function call.

If I include `word i; mem.first(i);` in my testbench:
```
cycle           1:
cycle           2:
cycle           3:
mi: ead08093, OP: (rf[ 1] = 00000000, rfn = fffffead) + 4294966957
mi: ead08093, OP: (rf[ 1] = fffffead, rfn = fffffd5a) + 4294966957
cycle           4:
cycle           5:
mi: f5610113, OP: (rf[ 2] = ff76e000, rfn = ff76df56) + 4294967126
mi: f5610113, OP: (rf[ 2] = ff76df56, rfn = ff76deac) + 4294967126
...
```

`mi` is read from the memory array `mem` and gets decoded. The circuit implements `rfn[x] = rf[x] + shown_integer` and  `rf[]` is updated using `rfn[]` in the next cycle, but the update happens in the same cycle before the values have settled (`ffff_fead` = `4294966957`). So the integer is added twice. (`4294966957 *2` = `ffff_fd5a`).

If I comment `mem.first(cycle);`:

```
cycle           1:
cycle           2:
cycle           3:
mi: ead08093, OP: (rf[ 1] = feedc000, rfn = feedbead) + 4294966957
cycle           4:
cycle           5:
mi: f5610113, OP: (rf[ 2] = ff76e000, rfn = ff76df56) + 4294967126
...
```
then the behavior is correct.

I have the feeling that an additional evaluation step is introduced if `first()` method is used but the registers get updated too early. What do you think?

I could not create a minimal design that shows this behavior in a manageable time. If a design would be helpful, I can happily create one.

- master: 5.003 devel rev v5.002-42-ge6da59eec
- Archlinux
- Fix assistance welcome

goekce, 2022-11-16:
I managed to create a smaller example:

```systemverilog
module alu_tb;

logic rst, clk = 0;
byte mi, mo, mi_addr, mi_addrn;
byte mem [byte];
byte rf, rfn;

integer cycle = 0;

initial begin
	mem[8'h80] = 8'hb7;
	mem[8'h81] = 8'hc0;
	mem[8'h82] = 8'hed;
	$dumpfile("signals.fst");
	$dumpvars();
	$monitor("t: %0d, mi= %h, rf= %h, rfn: %h, clk: %v, rst: %v", $time, mi, rf[1], rfn[1], clk, rst);
	//rst = 0; #1 rst = 1; #2 rst = 0;
end

always
	#1 clk = ~clk;

always @(posedge clk)
	begin
	cycle += 1;
	mi = mem[mi_addr];
	mem[0] = '1;
	end

always_comb begin
	case (cycle)
	1:
		rst = 1;
	2:
		rst = 0;

	5:
		$finish;

	100000:
		begin
		byte i;
		/* mem.first(i); */
		end
	endcase
end

always_ff @(posedge clk, posedge rst)
	begin
	$display("posedge begin");
	rf <= rst ? '0 : rfn;
	mi_addr <= rst ? 8'h80 : mi_addrn;
	end

always_comb begin
	mi_addrn = mi_addr;
	rfn = rf;

	if (rst == 0)
		begin
		mi_addrn = mi_addr + 1;
		
		case (mi)
		8'hb7:
			rfn = 5;

		8'hc0:
			rfn = rf + 1;
		default:
			rfn = rf;
		endcase
		end
end
endmodule
```
---
Waveform:
[image]

---
If I include `mem.first(i)`, then `rf` is updated in the same cycle:

[image]
---
I noticed that this behavior does not occur if I make `rst` independent of the `clk` (by using the `rst` driving line in the `initial` block). Additionally  deasserting `rst` on a clock edge could have also contributed to this problem.

I would like to understand what is happening or at least know if this is a feature or bug.

wsnyder, 2022-11-17:
Calling a function with side effects in an always_comb is prone to problems, as the always may be called multiple times as signals evaluate.  This should be moved to an always_ff or something otherwise triggered by an event.

What I believe is happening is .first() says that mem is both used and consumed and as such is causing the always_comb to be evaluated twice.  We can fix that for .first, but as above there could be other future Verilator changes, or changes in your code that would break your case again.

goekce, 2022-11-17:
> Calling a function with side effects in an always_comb is prone to problems, as the always may be called multiple times as signals evaluate. 

Why is this a problem?

> This should be moved to an always_ff or something otherwise triggered by an event.

You recommend that I should use impure functions only in a sequential block, right? My goal is to reuse combinational logic. How can I do that without functions in an `always_comb` block?

Thanks for your support Wilson.

> What I believe is happening is .first() says that mem is both used and consumed and as such is causing the always_comb to be evaluated twice

Is this the reason for `rfn = 07` in the second figure? In the first figure `rf` gets immediately the value of `rfn`, so I have the impression that `always_ff` is evaluated too early too.

wsnyder, 2022-11-17:
It's only a problem if the side effect breaks something, while unusual putting it in an always_comb isn't intrinsically wrong, just has potential to cause trouble.

In your particular example I'd suggest considering cycle in an always_ff block (like https://github.com/verilator/verilator/blob/master/test_regress/t/t_EXAMPLE.v

I'm not sure if that is related to rfn, it could be something else is going on, it would need some digging.

goekce, 2022-11-24:
> It's only a problem if the side effect breaks something, while unusual putting it in an always_comb isn't intrinsically wrong, just has potential to cause trouble.

Oh, I see. It is not about functions in general but putting functions like `mem.first()`.

I display now the values in the beginning of the `always_ff`:

<details><summary>Source file</summary>
<pre><code>module alu_tb;

logic rst, clk = 0;
byte mi, mo, mi_addr, mi_addrn;
byte mem [byte];
byte rf, rfn;

integer cycle = 0;

initial begin
	mem[8'h80] = 8'hb7;
	mem[8'h81] = 8'hc0;
	mem[8'h82] = 8'hed;
	$dumpfile("signals.fst");
	$dumpvars();
	$monitor("t: %0d, mi= %h, rf= %h, rfn: %h, clk: %v, rst: %v", $time, mi, rf, rfn, clk, rst);
	//rst = 0; #1 rst = 1; #2 rst = 0;
end

always
	#1 clk = ~clk;

always @(posedge clk)
	begin
	cycle += 1;
	mi = mem[mi_addr];
	mem[0] = '1;
	end

always_comb begin
	case (cycle)
	1:
		rst = 1;
	2:
		rst = 0;

	5:
		$finish;

	100000:
		begin
		byte i;
		/* mem.first(i); */
		end
	endcase
end

always_ff @(posedge clk, posedge rst)
	begin
	$display("[%0t] cycle=%0d, mi=%h, rf=%h, rfn:%h, clk:%v, rst:%v", $time, cycle, mi, rf, rfn, clk, rst);
	rf <= rst ? '0 : rfn;
	mi_addr <= rst ? 8'h80 : mi_addrn;
	end

always_comb begin
	mi_addrn = mi_addr;
	rfn = rf;

	if (rst == 0)
		begin
		mi_addrn = mi_addr + 1;
		
		case (mi)
		8'hb7:
			rfn = 5;

		8'hc0:
			rfn = rf + 1;
		default:
			rfn = rf;
		endcase
		end
end
endmodule
```
</code></pre>

The output without calling the function:

```
t: 0, mi_addr=00, mi=00, rf=00, rfn=00, clk=St0 , rst=St0 
t: 1, mi_addr=00, mi=00, rf=00, rfn=00, clk=St1 , rst=St0 
t: 1, mi_addr=00, mi=00, rf=00, rfn=00, clk=St1 , rst=St1 
[1] cycle=1, mi_addr=00, mi=00, rf=00, rfn:00, clk:St1 , rst:St1 
t: 1, mi_addr=80, mi=00, rf=00, rfn=00, clk=St1 , rst=St1 
t: 2, mi_addr=80, mi=00, rf=00, rfn=00, clk=St0 , rst=St1 
t: 2, mi_addr=80, mi=00, rf=00, rfn=00, clk=St0 , rst=St1 
t: 3, mi_addr=80, mi=00, rf=00, rfn=00, clk=St1 , rst=St1 
t: 3, mi_addr=80, mi=00, rf=00, rfn=00, clk=St1 , rst=St0 
[3] cycle=2, mi_addr=80, mi=00, rf=00, rfn:00, clk:St1 , rst:St0 
t: 3, mi_addr=81, mi=b7, rf=00, rfn=05, clk=St1 , rst=St0 
t: 4, mi_addr=81, mi=b7, rf=00, rfn=05, clk=St0 , rst=St0 
t: 4, mi_addr=81, mi=b7, rf=00, rfn=05, clk=St0 , rst=St0 
t: 5, mi_addr=81, mi=b7, rf=00, rfn=05, clk=St1 , rst=St0 
t: 5, mi_addr=81, mi=b7, rf=00, rfn=05, clk=St1 , rst=St0
```

Here the evaluation of the combinational signal `rfn` seems to be correct. If I include the function:

```
t: 0, mi_addr=00, mi=00, rf=00, rfn=00, clk=St0 , rst=St0 
t: 1, mi_addr=00, mi=00, rf=00, rfn=00, clk=St1 , rst=St0 
t: 1, mi_addr=00, mi=00, rf=00, rfn=00, clk=St1 , rst=St1 
[1] cycle=1, mi_addr=00, mi=00, rf=00, rfn:00, clk:St1 , rst:St1 
t: 1, mi_addr=80, mi=00, rf=00, rfn=00, clk=St1 , rst=St1 
t: 2, mi_addr=80, mi=00, rf=00, rfn=00, clk=St0 , rst=St1 
t: 2, mi_addr=80, mi=00, rf=00, rfn=00, clk=St0 , rst=St1 
t: 3, mi_addr=80, mi=00, rf=00, rfn=00, clk=St1 , rst=St1 
t: 3, mi_addr=80, mi=b7, rf=00, rfn=05, clk=St1 , rst=St0 
[3] cycle=2, mi_addr=80, mi=b7, rf=00, rfn:05, clk:St1 , rst:St0 
t: 3, mi_addr=81, mi=b7, rf=05, rfn=05, clk=St1 , rst=St0 
t: 4, mi_addr=81, mi=b7, rf=05, rfn=05, clk=St0 , rst=St0 
t: 4, mi_addr=81, mi=b7, rf=05, rfn=05, clk=St0 , rst=St0 
t: 5, mi_addr=81, mi=b7, rf=05, rfn=05, clk=St1 , rst=St0 
t: 5, mi_addr=81, mi=c0, rf=05, rfn=06, clk=St1 , rst=St0
...
```

The line `t: 3, mi_addr=80, mi=b7, rf=00, rfn=05, clk=St1 , rst=St0` before `[3]` looks suspicious to me. How can `mi` get updated with `mem[80]` before the `always_ff` block?

We have two `always` blocks clocked by `clk`. Probably their scheduling changes based on `mem.first()`. I also noticed that if I use a non-blocking assignment to `mi` (`mi <= mem[mi_addr]`, then the simulation behaves as expected regardless of the inclusion of `mem.first()`. Maybe a non-blocking assignment happens after all always blocks have been evaluated, but a blocking assignment happens immediately, so the value of `rfn` is based on the order of the `always` and `always_ff` block.

Does this make sense?

goekce, 2022-12-08:
There is no interest in this issue/question, so I lean towards closing the issue to open up space for more important issues.

I think the problem could be an inherent problem of the Systemverilog simulation itself.

closed 2022-12-08 by goekce