verilator/verilator #3757: ubuntu 22.04 missing zlibc

jjones-github, 2022-11-17:
Without `zlibc` in ubuntu 22.04 is it safe to install Verilator without it? The instructions say `sudo apt-get install zlibc zlib1g zlib1g-dev  # Ubuntu only (ignore if gives error) `. I assume if not available ignore installing it?

We are on a 22.04 image for other reasons and would love to keep Verilator on the same image so everyone can use the same image regardless of engineering discipline. 

I had to use `autoconf-archive` instead of `autoconf` on 22.04. Any idea which I should use with Verilator? 

In case it helps others, we built the latest SystemC and discovered that the archive was required to resolve an issue with `AC_PROG_LIBTOOL` not being found. So I ended up doing `apt purge -y autoconf && apt-get -y install autoconf-archive lib tool libltdl-dev` and then after pulling down SystemC inside the systemc directory `autoupdate && autoreconf -I` before configure and make.

**TL;DR**
A co-worker was chasing some odd behavior with our SystemC model using 2.3.3. We were using clang 12, tried g++, moved to ubuntu 22.04 from 20.04 but still was seeing odd behavior exiting a user function. Ended up building SystemC from [4a44d36](https://github.com/accellera-official/systemc/commit/4a44d3619eca48a7a99122ae8c0fe3baf101d2e5) after seeing valgrind errors on SystemC 2.3.3. This has is the latest currently and it is also valgrind clean.

wsnyder, 2022-11-17:
zlibc is needed for the Fst/GTKwave support at runtime, if you aren't using that you should be fine without it.

Thanks for the other tips.

closed 2022-11-17 by wsnyder

YiwenTang1029, 2022-12-21:
> zlibc is needed for the Fst/GTKwave support at runtime, if you aren't using that you should be fine without it.
> 
> Thanks for the other tips.

Can i ask how should i install the zlibc ? i used the command "sudo apt-get install zlibc" , it turns out "unable to locate the package zlibc". I've already installed the zlib1g and zlib1g-dev.

wsnyder, 2022-12-21:
Does the build fail?  If not you are done.  The packages installed for Ubuntu for the self-testing are listed in ci/ci-install.bash.