verilator/verilator #3758: Don't block Make jobserver when calling Verilator from a Makefile

gussmith23, 2022-11-18:
Hi all!

I run Verilator in parallel in a Makefile, but unfortunately, I think Verilator is blocking the Make jobserver from working:
```
make[1]: warning: -jN forced in submake: disabling jobserver mode.
```

I think this is because Verilator calls `$(MAKE)` with `-j1` in `Makefile.in`. I could be totally wrong, though. I don't fully understand how the Makefiles work.

Is there any way around this? It definitely slows down my build.

wsnyder, 2022-11-18:
Are you referring to building Verilator itself, or Verilated models with --build, or calling make yourself on a Verilated model?

gussmith23, 2022-11-18:
I just logged off my machine but tomorrow I'll post my makefile to show you
what I'm doing! I'm talking about verilating files.

gussmith23, 2022-11-18:
Here's my Makefile:

```
run_all: $(TESTBENCH_EXECUTABLES)

%.out: %.cc $(VERILATOR_INCLUDE_DIR)/verilated.cpp $(VERILOG_FILES:.v=.vo)
  # -lstdc++ fixes build problems on Mac.
  # The + passes information about the jobserver to sub-commands. Not sure if it has any effect here.
	+$(CXX) $(CFLAGS) -I$(VERILATOR_INCLUDE_DIR) -faligned-new -lstdc++ -std=c++11 -Wall -Wextra -Werror $^ -o $@
	$@ || (echo "Test failed: $@"; exit 1)

%.vo: %.v
  # The CFLAGS values fix issues with timing functions not being found.
  # The + passes information about the jobserver to sub-commands. Not sure if it has any effect here.
	+$(VERILATOR) -Wall --CFLAGS "-DVL_TIME_STAMP64 -DVL_NO_LEGACY" -Mdir . --cc --build $(VFLAGS) $<
  # Copy the compiled result, which is named V<filename>__ALL.o, to <filename>.vo.
	cp $(addprefix $(dir $<)/V, $(patsubst %.v,%__ALL.o,$(notdir $<))) $@
```

wsnyder, 2022-11-18:
Try "verilator --build-jobs 0".

wsnyder, 2022-11-18:
Oh, actually that will set -j to the CPU count.  I think we lost the ability to have no -j in the make.  I'm not sure what should be the behavior, as most people are not running a job server, so we defaulted to something that seemed more common (for better and worse).  Any suggestions?

wsnyder, 2022-11-19:
One idea is if MAKEFLAGS environment contains --jobserver-auth then skip passing a -j.

gezalore, 2022-11-19:
Worst case could add a special case of "-j none" if we want to get back the "no -j passed to make" behaviour. Can be considered a hack though.

closed 2022-12-11 by wsnyder

gussmith23, 2022-12-12:
Thanks so much Wilson, you rock! I expect this to speed up my tests a lot!