verilator/verilator #3761: Parameter enum casting incorrectly throws error

AndrewNolte, 2022-11-18:
https://github.com/verilator/verilator/commit/d25834e57b77fd9712628143646ddd82a8de3c1d
 recently introduced this check

## Test Case
```
module t;

   typedef enum {ZERO, ONE, TWO} e_t;

   localparam e_t param_enum = e_t'(0);

endmodule
```

produces

```
======================================================================
vlt/t_enum_type_elaborate: ==================================================
        perl /usr/scratch/anolte/local/verilator/test_regress/../bin/verilator --prefix Vt_enum_type_elaborate --x-assign unique -cc -Mdir obj_vlt/t_enum_type_elaborate --fdedup --debug-check --comp-limit-members 10 --lint-only --clk clk  -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_enum_type_elaborate t/t_enum_type_elaborate.v    > obj_vlt/t_enum_type_elaborate/vlt_compile.log
%Error-ENUMVALUE: t/t_enum_type_elaborate.v:11:19: Illegal implicit conversion to enum 't.e_t' from 'logic[31:0]' (IEEE 1800-2017 6.19.3)
                                                 : ... In instance t
                                                 : ... Suggest use enum's mnemonic, or static cast
   11 |    localparam e_t param_enum = e_t'(0);
      |                   ^~~~~~~~~~
                  ... For error description see https://verilator.org/warn/ENUMVALUE?v=5.003
%Error: Exiting due to 1 error(s)
%Warning: vlt/t_enum_type_elaborate: Exec of perl failed: %Error-ENUMVALUE: t/t_enum_type_elaborate.v:11:19: Illegal implicit conversion to enum 't.e_t' from 'logic[31:0]' (IEEE 1800-2017 6.19.3)

vlt/t_enum_type_elaborate: %Error: Exec of perl failed: %Error-ENUMVALUE: t/t_enum_type_elaborate.v:11:19: Illegal implicit conversion to enum 't.e_t' from 'logic[31:0]' (IEEE 1800-2017 6.19.3)
vlt/t_enum_type_elaborate: FAILED: Exec of perl failed: %Error-ENUMVALUE: t/t_enum_type_elaborate.v:11:19: Illegal implicit conversion to enum 't.e_t' from 'logic[31:0]' (IEEE 1800-2017 6.19.3)
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:00

======================================================================
```

wsnyder, 2022-11-19:
Thanks for the report, should be good now.

wsnyder, 2022-11-19:
Missed part of the commit, stay tuned...

wsnyder, 2022-11-19:
Should be good now, thanks for the good test.

closed 2022-11-19 by wsnyder