verilator/verilator #3770: Rare clock skew-like issue

flaviens, 2022-11-21:
Hi there!

I encountered a surprising issue where the Verilator (`Verilator 5.003 devel rev v4.228-192-g79682e607`) output does not comply with the SystemVerilog input.
[To help you reproduce the issue, I created a repository here](https://github.com/flaviens/veriskew).
In short, according to the snippet below, the signal `signal_1154` should follow `signal_0672`, but under some circumstances, they both take the same value simultaneously.

```
  always_ff @(posedge clk_i, negedge rst_ni)
    if (!rst_ni) signal_1154 <= 64'h0000000000000000;
    else signal_1154 <= signal_0672;
```

I analyzed a bit the C++ code generated by Verilator and it seems that the cause is a wrong evaluation order during the clock positive edge, where first `signal_0672` takes its new value, and this new value propagates directly to `signal_1154` during the same `eval()` call, but my analysis may be erroneous.

If you have any question please feel free to ask!

Thank you for your help!
Flavien

wsnyder, 2022-11-21:
Thanks for providing an example, however 8MB is too much to debug, please reduce the test case to the minimum possible to show the issue. Thanks.

wsnyder, 2022-11-22:
That's still hundreds of KB, sorry but I am unable to spend a lot of time on debugging code this large.  Please get it down to something that is truly minimal, that is you can't remove a single verilog/C++ line without it being critical to seeing the bug.

flaviens, 2022-11-29:
Hi @wsnyder, thank you for your feedback!
* Unfortunately whenever I reduce the amount of SystemVerilog code substantially, the issue disappears. The problem  looks quite subtle.
* To be certain that I was not doing a mistake in my waveform analysis, I tried exactly the same inputs on the same hardware source in a commercial simulator. It confirms that the results differ between Verilator and the commercial simulator.
Despite the codebase that I provided looking big, its structure is quite straightforward and I pinpointed where the issue occurs. Therefore, there is hope that the suspected Verilator bug is easy to identify for someone familiar with the Verilator source.
May you need any support, I remain at your disposal.

wsnyder, 2022-12-02:
Please decouple this from fusesoc, I need just a test that uses Verilator only.

I'm suspicious of the /*verilator public*/ comment on the signal that is causing problems.

Also please try -fno-inline.  I suspect that will let you make a smaller example (likely once it gets small the inliner takes over and hides the bug.)

Alternatively, please try to debug Verilator yourself.  Run with --debug for a good (e.g. the inlined case) and bad case, look for the signal in the .tree files, and see if you can determine where it goes bad.

flaviens, 2022-12-02:
Hello @wsnyder,

Thank you very much for your response!

- I tried without any /* verilator public */, unfortunately it does not make any difference.
- I tried -fno-inline, no difference as well.
- I now decoupled from fusesoc as you requested (without impact on the bug).

After a lot of trial and error, I could this time significantly reduce the codebase while maintaining the bug. [Please find the reduced version here](https://github.com/flaviens/veriskewsmall).

I remain at your disposal as always.
Thanks!

wsnyder, 2022-12-03:
Here's the "real" minimal example ;)

```
module wt_axi_adapter (
  input clk_i /* verilator public */,
  input rst_ni /* verilator public */
                       );
wire [63:0] signal_0305;
wire [63:0] signal_0306;
wire [63:0] signal_0307;
wire [63:0] signal_0671;
wire [63:0] signal_0672 /*verilator public*/;
wire [63:0] axi_rd_data /* verilator public */;
wire [63:0] axi_rd_data_t0 /* verilator public */;
wire [3:0] axi_rd_id_out /* verilator public */;
wire [3:0] axi_rd_id_out_t0 /* verilator public */;
wire axi_rd_last /* verilator public */;
wire axi_rd_last_t0 /* verilator public */;
wire axi_rd_valid /* verilator public */;
wire invalidate /* verilator public */;
  assign signal_0305 = { { 64{ icache_rtrn_rd_en } } } & axi_rd_data_t0;
  assign signal_0671 = signal_0305 | signal_0306;
  assign signal_0672 = signal_0671 | signal_0307;
reg [63:0] signal_1154 /* verilator public */;
  always_ff @(posedge clk_i, negedge rst_ni)
    if (!rst_ni) signal_1154 <= 64'h0000000000000000;
    else signal_1154 <= signal_0672;
  assign axi_rd_rdy = ~ invalidate;
  assign signal_0976 = ! axi_rd_id_out;
  assign signal_0978 = axi_rd_valid && signal_0976;
  assign signal_0980 = signal_0978 && axi_rd_rdy;
  assign icache_rtrn_rd_en = signal_0980 ? 1'h1 : 1'h0;
endmodule
```

Anyhow the problem is the public variables.  Public is intended to allow reading of sequential state, but it does not change the scheduling (it isn't a "force"). So your testbench that forces the values is causing the problem.    You should change the forced states to inputs and not use public at all, or at least change flops.

flaviens, 2022-12-03:
Hi @wsnyder,

Thank you for taking the time for analyzing this case! It is very useful.
Unfortunately the problem is not as simple.
[Please find in the updated repo here a similar occurrence of the bug, where no signal is annotated as _verilator public_.](https://github.com/flaviens/veriskewsmall)

As always, I'm at your disposal.
Thanks!

wsnyder, 2022-12-03:
You are applying the input data changes the same time as changing the clock.  It is undefined which ordering you will get, that is if the data supposed to be observed in the previous or next cycle. If I change your recording.txt to set the data in one line, then apply the clock separately in the next line, then I do not see the problem.

flaviens, 2022-12-03:
Hi @wsnyder,

Thank you for analyzing this!
Your observation is correct. However this is quite a bad news. When reducing the design, the symptoms were reproduced but not the root cause.

We still do not know the root cause of the original problem as described at the beginning of the thread ([see the updated original repo](https://github.com/flaviens/veriskew)). The original design really does nothing crazy:
- it does not act on any design input except the reset and clock signals.
- it does not read or write any internal signal (except the SRAM initially through traditional DPI, which had never been a problem so far).

We saw that the symptom is also a bit divergent btw (which should have warned me that the reduction wasn't great): in the original issue, the flip-flop is _sometimes_ skipped, whereas in the reduced versions it was always skipped, see the waves of the original problem below.

[image]

So there's still something wrong and yet unexplained. My guess is that there is a faulty edge case in Verilator that we are hitting. How do you suggest to proceed next? Unfortunately so far, all attempts to reduce the design and keep the root cause failed.

I remain at your disposal as always.
Thanks!

wsnyder, 2022-12-04:
First, make the test self checking so it fails with an obvious message only when that bad case occurs, and not if something else goes wrong.  Then start deleting code making sure that the message stays the same.  That's how I reduced the other cases you sent.

flaviens, 2022-12-04:
Hi @wsnyder,
Thank you for the suggestions! I could only remove one level of hierarchy (the topmost). Whenever I remove another level, the bug disappears.

* [This branch](https://github.com/flaviens/veriskewsmall) contains the smallest design I got with the bug still happening.
* [Here](https://github.com/flaviens/veriskewsmall/tree/cva6_top_nobug) is the largest with the bug not happening.

How do you envision the next steps?

As always, I'm at your disposal.
Thanks!

wsnyder, 2022-12-06:
Try the -fno-inline to see if that lets you make the test case smaller.

Other than that you'll need to look at the .tree files to see how the code in question is getting misscheduled or otherwise causing the bug. This will be difficult to do/understand which is why a smaller test case is advisable.

flaviens, 2022-12-06:
Hi @wsnyder, thank you for the tip! [The flag is a bit hidden but it's unfortunately already here](https://github.com/flaviens/veriskewsmall/blob/7b833ddba370b768e5e9ef2890bce8214b97484f/build_test.sh#L39).

What is your next advice?
Thanks!

wsnyder, 2022-12-08:
You will need to dig into the tree files to determine what is misordered and determine the source fixes.

flaviens, 2022-12-08:
Hi @wsnyder,

Thank you for your response!
I seems however that for doing that, we need a Verilator developer, not a Verilator user.

flaviens, 2022-12-09:
Hello!

Suprise: I encountered the same bug on another design, and fortunately it is considerably smaller than the aforementioned design.
Again in this bug, the output of the register does not follow the flop-flop's input correctly, but if isolate down some levels of hierarchy, then the register behaves correctly.

In short, we have this register in the source
  always_ff @(posedge clk_i)
    instr_rdata_c_id_o_t0 <= signal124;

[image]

* [Here](https://github.com/flaviens/veriskewsmall-ibex/tree/record_memtop) is the smallest version of the new design I got with the bug still happening.
* [Here](https://github.com/flaviens/veriskewsmall-ibex/tree/record_rvcore) is the largest version of the new design I got with the bug not visible anymore, however the invisibility of the bug here may be due to the isolation process of the module. [When going down by one module more](https://github.com/flaviens/veriskewsmall-ibex/tree/record_ucore), the register clearly works correctly.

Maybe this simpler design will help debugging this issue. It also shows that the Verilator bug is not specific to the design mentioned above.

wsnyder, 2022-12-09:
The "smaller" design is 1.6 MB, and seems to still have a lot of stuff that I suspect is unnecessary to show the issue.  Am I looking at the wrong thing?

flaviens, 2022-12-09:
Hi @wsnyder, yes you're probably looking at the right thing! What _unneccessary_ things for example?

wsnyder, 2022-12-09:
The very first structure in the file for example crash_dump_t is never used, so I believe (but didn't prove) it can be removed.  Therefore I don't think you minimized the test.

What I want is a test that is:
1. Uses only Verilator no other tools
2. Self testing, in that it fails if the problem occurs in a unique way (with a message) and succeeds otherwise.
3. All unnecessary stimulus/checking has been removed (e.g. don't check signals that don't cause the problem).
4. All code that is not necessary to show the unique failure message has been removed.
5. Is ~~10KB or less.   Usually disabling inlining makes this possible.

Alternatively, and preferable, is to dig into the Verilator debug dumps yourselves or from a commercial support firm, and debug it ;)

flaviens, 2022-12-09:
Requirement 1 is already satisfied if I'm not mistaken.
Requirement 2 is for the moment an immediate clear waveform, but can be made into a printed message if we get the other requirements satisfied. Not a problem.
Requirement 5 is unlikely to happen, given the experience we have above with both designs where the bug disappears whenever we move non-trivially down into the module hierarchy. For sure we could remove some unused parameters and types in packages (packages are only used for the very top-level modules), but it will most likely not make a useful difference.

flaviens, 2023-01-16:
Hello! For informational purposes, is there any plan to fix the issue?
Thanks!
Flavien

wsnyder, 2023-01-16:
There is currently too big a backlog for me to personally debug  such a large case again this at this time, it will be gotten to eventually. If it's a priority to you, your looking at is the fastest path to a fix.

flaviens, 2023-01-16:
Thank you for your response! That sounds reasonable. I'll post here if there are any updates on my side.

wsnyder, 2025-03-31:
Never received a minimized test, if there is one or you make one, please open a new issue.