verilator/verilator #3783: Incorrect Error-ENUMVALUE when instantiating a module

furkanusta, 2022-11-26:
Test code:

```verilog
typedef enum logic [1:0] {A, B, C, D } letters;

module my_module
 #(parameter letters LETTER = letters'(0)) 
  ();
endmodule
    
module tb ();
    my_module #(.LETTER(letters'(0))) dut ();
endmodule    

```

I am using (4b3731d31, current master)
```bash
> verilator --version
Verilator 5.003 devel rev v5.002-91-g4b3731d31

> verilator --lint-only --top-module tb bug.sv 
%Error-ENUMVALUE: bug.sv:4:22: Implicit conversion to enum '__024unit::letters' from 'logic[1:0]' (IEEE 1800-2017 6.19.3)
                                : ... In instance tb.dut
                                : ... Suggest use enum's mnemonic, or static cast
    4 |  #(parameter letters LETTER = letters'(0)) 
      |                      ^~~~~~
                  ... For error description see https://verilator.org/warn/ENUMVALUE?v=5.003
```

Also it would be nice to have if the error showed the location of instantiation instead of definition

wsnyder, 2022-12-11:
Have been working on this for a while, lots of edge cases.  Note you can use -Wno-ENUMVALUE to get around this.

wsnyder, 2022-12-11:
Fixed in

closed 2022-12-11 by wsnyder

furkanusta, 2022-12-12:
The test case I initially pasted passes but this one fails (extracted from the same file)

```verilog
typedef enum logic [2:0] {A, B, C, D } enum_test;

function automatic enum_test foo(int a);
    return enum_test'(1);
endfunction

module tb ();
    localparam FMT = foo(1);
endmodule

```  
I am using the latest commit 01b521a