verilator/verilator #3786: %Error: Internal Error: ../V3Const.cpp:135: Bad word index into wide

kanedayo, 2022-11-29:
Thanks to providing most useful tool ever.
btw, I've encountered some mysterious error.

Can you attach an example that shows the issue?
```
// verilator -Wall --lint-only  sample.sv -GN=6 # OK
// verilator -Wall --lint-only  sample.sv -GN=7 # NG

module sample
  #( parameter N = 4 )
  ( input [2**N-1:0] a,b, output c );

  assign c = ^{a,b}; // OK@N=6, NG@N=7

endmodule
```

What 'verilator --version' are you using?  Did you try it with the git master version?
```
$ verilator --version
Verilator 5.003 devel rev v5.002-102-g68e1b473e
```

What OS and distribution are you using?
```
$ cat /etc/redhat-release
Rocky Linux release 8.6 (Green Obsidian)
```

wsnyder, 2022-11-29:
@yTakatsukasa might you be able to =take a look? It's in the V3Const bit optimizer code you added.  I don't see anything obviously wrong in the tree before the optimization starts.  Note I just pushed some UASSERT updates to make it print the node involved.

Note the bug only shows up with --lint-only, not -cc, so probably requires some pattern that usually gets optimized away before this code sees it.

yTakatsukasa, 2022-11-30:
Thanks for letting me know. will take a look at this.

yTakatsukasa, 2022-12-01:
Now the master has the fix.
Thanks for the report with concise test case.

kanedayo, 2022-12-05:
Thank you very much. It works well at the version below, and the flag (-fno-const-bit-op-tree) also work around at the old version.
```
$ verilator --version
Verilator 5.003 devel rev v5.002-127-gea1b141d1
```

PR #3793 (2022-12-01): Fix 3786 (Internal error in bit op tree optimization)
fixes #3786 

When `--lint-only` is set,V3Expand, that inserts WORDSEL, is skipped.
Bit op tree optimization uses uint64_t for constants (such as bit mask).
This PR disables opt. for such wide number when WORDSEL is missing.

As usual, I push a test that reproduces the issue, then push the fix later.