verilator/verilator #3787: Performance analysis and execution hardware questions

shareefj, 2022-11-29:
Hi there,

we're using Verilator for software bringup and although it's working nicely, we've never really looked at how the model is performing.  So there are two parts to this:

1.  Are there any recommendations for the best hardware to use?  I remember seeing a post somewhere that stated CPUs with large caches are a benefit but has anyone ever looked at this is more detail?  With the latest AMD releases, are there any recommendations for fast exec hosts?

2. We've noticed that in some cases running multi-threaded is slower than running the model single threaded.  Where do we start if we want to try and look at the performance of the model?  Are there any guides out there for getting the best out of Verilator?

Thanks, Shareef.

gezalore, 2022-11-29:
I would recommend starting with https://verilator.org/guide/latest/simulating.html which answers some of these questions and provides more performance related information.

wsnyder, 2022-11-29:
I haven't seen recent benchmark results but generally the larger the cache and closer to the CPU it is, the better performing.  Traditionally this favors the server-class Intel CPUs, but mileage varies, if you get data please consider sharing it.

Closing as answered, but feel free to follow up here.

closed 2022-11-29 by wsnyder