verilator/verilator #3791: Going beyond IEEE standard - support call of python function like "DPI-C" in SystemVerilog from Verilator

vishalkatigar, 2022-11-30:
First of all, I would like to thank all who contributed to Verilator...🙏

I am a verification engineer who comes up with just ideas, I am not the right person to decide if it is possible or viable.

what if we can have feature in Verilator to call the python function from SystemVerilog code the same as the "DPI-C" function call we are doing for communication between SV and C language?

it might be compiled python code or it might be compiled when we invoke Verilator.

Requesting all developers of Verilator to discuss and add if possible.

wsnyder, 2022-11-30:
Perhaps you are looking for https://www.cocotb.org/ that already supports Verilator?

vishalkatigar, 2022-11-30:
Hi Wilson,

No, I know Cocotb but it is a complex software layer on top of the Verilator with the use of VPI or VHPI.

also, we may have a bug in Cocotb sometime (I know now it is very stable, but new feature development may lead to any unknown bug)

we don't want to implement everything in python we can use python for stimulus generation or scoreboard/reference model development (as we have a wide availability of libraries for everything).

So what I am asking in terms of advanced features is just we can call any python function from Systermverilog language itself (which is not part of IEEE or SystemVerilog LRM) without depending on any additional software layer i.e. Cocotb

please follow below code snippets to see what kind of feature I am thinking of.

in SystemVerilog
```
  `include "verilator_py.sv"
  //export and import for a function call from python to SV and vice-versa
  export "DPI-PY" function sv_method;
  import "DPI-PY" python_method();

  function new();
     // Calling function implemented in python using "DPI-PY"
     python_function_call_from_sv();
  endfunction

  function void sv_method();
    $display("Function call from python...!");
  endfunction
```
in python
```
  #Not accurate representation just trying to give example here
  import XXXX # importing other modules
  from Verilator import svdpi

  #function call from SV
  def python_function_call_from_sv():
        print("Python function called from SV")

  #Calling SV function from python
  svdpi.sv_method();
```
Hope you understand the use case.

The disadvantage of verification in pure python:
- Slow down the simulation.
- At SoC/Chip level simulations are not possible due to slowness.
- we don't want to implement coverage in Cocotb (additional packages are required to be imported) if that feature is already available in Verilator.
- SV Assertions are not possible in Cocotb.
- likewise, we have many more reasons we can not switch to full python verification in the current situation.

Advantages of using python alongside SV :
- Increase productivity to generate stimulus and create a reference model/scoreboard
- Reduce time as less python code is used (maybe only 20 or 30 function calls to python are required)
- Easy and fast adoption for old as well as for new engineers, because no burden to learn Cocotb or complete python, basic python is enough.
- Reusing existing components(SV/UVM) or developing low-level components like the driver and monitor are easy in SV than in Python.
- We can use rich functionality like SV/UVM for coverage, assertion, mod port, clocking blocks, interfaces, etc.
- We don't have to worry about the bug in Cocotb or the dependency module.
- We are not getting blocked because some features are not available in Cocotb or a bug needs to be fixed.
- Increase Adoption of Verilator as the time to market gets reduced due to increased productivity. (Academia/Industry prefers to use first)
- As more students/companies get involved to use Verilator, we may get more contributors 👍.

wsnyder, 2022-11-30:
I understand your points, but it still seems that you are recreating several pieces of CocoTB, I'd encourage you to look at it closely.

At this time I don't think I would accept a patch to add an integrated python interface in Verilator, even if you or someone does all the plumbing, as the long term support will be too much work.  Verilator at one point had connections to Perl for preprocessing etc, and it was extremely difficult to maintain, and that did a very small subset of what would be required (it didn't even handle wide data types.)

I'd suggest you consider this python interface as a separate external package that hooks up through the normal C-DPI interface.  You can use the Verilator XML interface to get information about the design.  If there are improvements or hooks needed in Verilator to support such an external tool they'll certainly be considered and likely accepted if reasonable.

Closing as answered but feel free to follow up.

closed 2022-11-30 by wsnyder