verilator/verilator #3799: Wrong output to text file

jotego, 2022-12-04:
The following code:

```
integer fdebug=0, line_cnt=0;

initial begin
    if( fdebug==0 )
        fdebug=$fopen("main_io.log","w");
end

always @(posedge clk) begin
    if( rst ) begin
        line_cnt <= 0;
    end else if(cen6) begin
        if( ram_cs ) line_cnt <= line_cnt+1;
        if( ram_cs &&  cpu_rnw ) $fdisplay(fdebug,"%04X -> %02X - %d", A, ram_dout, line_cnt );
        if( ram_cs && !cpu_rnw ) $fdisplay(fdebug,"%04X <- %02X - %d", A, dout, line_cnt );
    end
end
```
Should produce a distinct line_cnt output for each line in the _main_io.log_ text file. In Modelsim, that's the case. But somehow in Verilator you get the first four lines printed three times:

```
effe <- 01 -           0
effd <- ac -           1
effd -> ac -           2
effe -> 01 -           3
effe <- 01 -           0
effd <- ac -           1
effd -> ac -           2
effe -> 01 -           3
effe <- 01 -           0
effd <- ac -           1
effd -> ac -           2
effe -> 01 -           3
e000 -> 00 -           4
e000 <- 55 -           5
e000 -> 55 -           6
e000 <- 00 -           7
e001 -> 00 -           8
```
Note that the number in the last column should always increase. Note that **the waveforms are correct** when checked in gtkwave. The waveforms do follow the expected behaviour. The problem seems to be related to the file output. The correct behavior is:

```
effe <- 01 -           0
effd <- ac -           1
effd -> ac -           2
effe -> 01 -           3
e000 -> 00 -           4
e000 <- 55 -           5
e000 -> 55 -           6
e000 <- 00 -           7
e001 -> 00 -           8
e001 <- 55 -           9
```

**_Can you attach an example that shows the issue?_**  
It's a convoluted setup. Let me just explain the issue in case something similar shows up.

**_What 'verilator --version' are you using?  Did you try it with the git master version?_**
Verilator 5.002 2022-10-29 rev v5.002

**_What OS and distribution are you using?_**
Linux once 5.4.0-132-generic #148-Ubuntu SMP Mon Oct 17 16:02:06 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

**_May we assist you in trying to fix this in Verilator yourself?_**
Let me assist only in providing bug information rather than software development

wsnyder, 2022-12-04:
If you can make a testcase should be easy to debug.  Alternatively:

Are you using multithreading? If so try one thread.

Try -fno-split.  Try the other -fno options described in the manual.

jotego, 2022-12-05:
Thank you for the hint. I am not using multithreading because that option runs considerably slower in all the computers I have had in the last four years. But that's a whole different topic.

I have tried all the -fno options using a small script:

```
for i in acyc-simp assemble case combine const const-bit-op-tree \
    dedup dfg dfg-peephole dfg-pre-inline dfg-post-inline expand \
    gate inline life life-post localize merge-cond merge-cond-motion \
    merge-const-pool reloop reorder split subst subst-const table; do
    sim.sh -video 2 -d VERILATOR_KEEP_CPU -d RST_DLY=9000 -args -fno-$i;
    mv main_io.log main_io_$i.log
done

for i in main_io*.log; do
    echo $i
    cat $i
done
```

They all produce the same wrong result 😭

jotego, 2022-12-05:
The problem happens in the regular $display calls too, without the need of involving a file.

I found what triggers it. The C++ file that wraps my simulation outputs information using **cout**, like this:

```
cout << ".";
cout.flush();
```
This is done every 16ms. Note that:

- Removing the flush call does not solve the problem
- Changing cout for cerr does not solve it

But if I remove the above two statements and replace them with `fputc('.', stderr);`, the problem goes away. So there is some interaction between the C++ I/O system here. I can get by with the C functions in my current work. Please feel free to close this issue but maybe this is something you want to explore as it should be ok to print something on screen without breaking the verilog $display calls.

jotego, 2022-12-05:
Sorry but the weird situation actually got weirder after the change. When I use the C++ library and I redirect the simulation output to a file I get the following. Note how this part 
```M/effe <- 01/ 
M/effd <- ac/ 
M/effd -> ac/ 
M/effe -> 01/
```
is repeated twice in here:

```
Verilator sim starts
Multibank SDRAM enabled
Read 800000 from sdram_bank0.bin
Read 800000 from sdram_bank1.bin
Read 800000 from sdram_bank2.bin
Read 800000 from sdram_bank3.bin
Read 150000 bytes from rom.bin
Simulation clock period set to 20832 ps (48.003072 MHz) INFO: Read    vram_lo.bin (4096 bytes/100%) for TOP.game_test.u_game.u_game.u_video.u_gfx.u_vram.u_lo.u_ram.genblk1
INFO: Read    vram_hi.bin (4096 bytes/100%) for TOP.game_test.u_game.u_game.u_video.u_gfx.u_vram.u_hi.u_ram.genblk1
INFO: Read        col.bin (1024 bytes/100%) for TOP.game_test.u_game.u_game.u_video.u_gfx.u_yram.u_ram.genblk1
INFO: Read        pal.bin (1024 bytes/100%) for TOP.game_test.u_game.u_game.u_video.u_colmix.u_comm.u_ram.genblk1
WARNING: Using Verilog version of T80 for simulation.
WARNING: Using Verilog version of T80 for simulation.
SDRAM burst mode changed to 2 mask 0xfffffffe
ROM download shortened to 32 bytes
ROM file transfered
.M/effe <- 01/
M/effd <- ac/
M/effd -> ac/
M/effe -> 01/
M/effe <- 01/
M/effd <- ac/
M/effd -> ac/
M/effe -> 01/
...M/e000 -> 00/
M/e000 <- 55/
M/e000 -> 55/
M/e000 <- 00/
M/e001 -> 00/
```
When moving to the C library functions, I thought it was good but what actually happened is that a larger section of the output gets copied twice. The weirdest thing is that this only seems to happen when I redirect the output to a file.

```Simulation clock period set to 20832 ps (48.003072 MHz) INFO: Read    vram_lo.bin (4096 bytes/100%) for TOP.game_test.u_game.u_game.u_video.u_gfx.u_vram.u_lo.u_ram.genblk1
INFO: Read    vram_hi.bin (4096 bytes/100%) for TOP.game_test.u_game.u_game.u_video.u_gfx.u_vram.u_hi.u_ram.genblk1
INFO: Read        col.bin (1024 bytes/100%) for TOP.game_test.u_game.u_game.u_video.u_gfx.u_yram.u_ram.genblk1
INFO: Read        pal.bin (1024 bytes/100%) for TOP.game_test.u_game.u_game.u_video.u_colmix.u_comm.u_ram.genblk1
WARNING: Using Verilog version of T80 for simulation.
WARNING: Using Verilog version of T80 for simulation.
M/effe <- 01/
M/effd <- ac/
M/effd -> ac/
M/effe -> 01/
Simulation clock period set to 20832 ps (48.003072 MHz) INFO: Read    vram_lo.bin (4096 bytes/100%) for TOP.game_test.u_game.u_game.u_video.u_gfx.u_vram.u_lo.u_ram.genblk1
INFO: Read    vram_hi.bin (4096 bytes/100%) for TOP.game_test.u_game.u_game.u_video.u_gfx.u_vram.u_hi.u_ram.genblk1
INFO: Read        col.bin (1024 bytes/100%) for TOP.game_test.u_game.u_game.u_video.u_gfx.u_yram.u_ram.genblk1
INFO: Read        pal.bin (1024 bytes/100%) for TOP.game_test.u_game.u_game.u_video.u_colmix.u_comm.u_ram.genblk1
WARNING: Using Verilog version of T80 for simulation.
WARNING: Using Verilog version of T80 for simulation.
M/effe <- 01/
M/effd <- ac/
M/effd -> ac/
M/effe -> 01/
M/e000 -> 00/
M/e000 <- 55/
M/e000 -> 55/
M/e000 <- 00/
M/e001 -> 00/
M/e001 <- 55/
M/e001 -> 55/
```
I don't know anymore whether this is a problem with bash, verilator, C++ or computing life in general!

wsnyder, 2022-12-05:
Is the problem there in 4.228?

Are you overriding VL_PRINTF?

Note Verilator does its own buffering in that VL_PRINTF_MT adds to a queue which later gets output with VL_PRINTF, but I can't explain why that would ever duplicate.

jotego, 2022-12-05:
I'll check version 4.228.

It looks like Verilator is corrupting the information passed to `$display` too. The `$display` statements produce these lines:

```
  1686  d000 <- aa
  1687  d000 -> aa
  1688  d000 <- 55
  1689  d000 -> 55
  1690  d000 -> aa     // Wrong arrow sequence, only one line with aa
  1691  d000 <- 55
  1692  d000 -> 55
  1693  d000 <- aa
  1694  d000 -> aa
```

Note how you get two _aa_, then two _55_, but then a single _aa_. The funny thing is that if you check out the waveforms, that's not what you get. The waveforms have the right sequence of two _aa_, then two _55_. Also the arrows should go left and right each time, but right where the orphan _aa_ is, the arrow sequence is lost too. Again, the waveform signal represented by the arrows is correct. Somehow the `$display` task produces a wrong output.

Oddly enough, the numeric sequence in the first column is not broken. It is as though Verilator was messing up the time at which the signals going into `$display` are read.

The same simulation setup in iVerilog or Modelsim produces the right text file.

jotego, 2022-12-05:
I think the _aa_ - _55_ thing may have been a bug in my data post-processing.  So we're down to the text buffer duplication which does not occur in 4.228 but it occurs in 5.002

wsnyder, 2022-12-11:
We'll need a self contained test case that shows the issue, or you will need to debug it yourself.  I would suggest look into how the message queue works around VL_PRINTF_MT.  There is presumably something different about your setup or code to hit this problem.

jotego, 2022-12-12:
I cannot provide a self contained test case, sorry. The test bench executes software in a CPU and that software is proprietary and I do not have the legal right to share them.
My work around this issue was to parse the VCD file instead of relying on the _$display_ task.
I will close the issue for now. If I come up again with this problem in a test bench I fully own, I will share it with you. Thank you very much for your kind help.

jotego, 2023-05-31:
This happened again on `Verilator 5.006 2023-01-22 rev v5.006` on a different test bench. There is a CPU writting data to a memory. 

The memory dumps the writes to the screen:
```
reg [AW:1] al;
reg [15:0] dl;
reg [ 1:0] wel;
always @(posedge clk) begin
    al  <= addr;
    dl  <= data;
    wel <= we;
    if( al!=addr || dl!=data || wel!=we ) begin
        if(we[0]) $display("%m %0X=%X", { {32-AW{1'b0}}, addr,1'b0}+VERBOSE_OFFSET,data[7:0]);
        if(we[1]) $display("%m %0X=%X", { {32-AW{1'b0}}, addr,1'b1}+VERBOSE_OFFSET,data[15:8]);
    end
end
```
Eventually the `$display` statements will contain data not present in the waveforms. In this compare file, the left column comes from an emulator and the right from **Verilator**.  The `>` disagreements shown are not real when checked against the waveform dump.

```
 64e6=ff							 64e6=ff
 6bf8=e1						      |	 6bf8=dc
							      >	 6bf9=2f
							      >	 6bfa=ff
							      >	 6bfb=00
							      >	 6bf6=82
							      >	 6bf7=8b
							      >	 6660=00
							      >	 64e6=ff
							      >	 6bf8=dc
 6bf9=2f							 6bf9=2f
 6bfa=ff							 6bfa=ff
```

The problem is thus still there and it is still an ugly bug. My simulation set up is a bit convoluted because of multiple git repositories, environment variables, etc. If I set up a docker container with it, would you look at this?

wsnyder, 2023-06-01:
I'm sorry but it is unlikely someone will have the time to debug your issue for you. One alternative is to contact one of the companies providing professional support: https://veripool.org/verilator_commercial_support/

wsnyder, 2023-07-02:
Closing until get self-contained example.

closed 2023-07-02 by wsnyder

jotego, 2023-07-03:
I'll try to prepare one. I am not sure how large the simulation needs to be in order to see the problem. I am getting around by printing the line number from a counter in Verilog, and then filtering and sortening based on that counter.

About the commercial support, I tried contacting with those companies in the past but didn't get a response. I wonder if the still provide a service.