verilator/verilator #3807: Case statements sometimes generate duplicated if conditions

AndrewNolte, 2022-12-09:
Example here https://github.com/verilator/verilator/pull/3806/files

It would also be nice if Verilator has a similar error/warning for verilog branches, this was a cpp warning though.

ArturBieniek4, 2026-09-02:
Fixed in https://github.com/verilator/verilator/pull/6495.

closed 2026-09-02 by ArturBieniek4

wsnyder, 2026-09-02:
@ArturBieniek4  the correct way to track fixed-in is to go to the original bug (#6945) and edit the comment to say "Fixed in #3807" in the *top* message.  It will then track properly, I did this and see how at top of #3807 it now has the merge commit listed.

ArturBieniek4, 2026-09-02:
> [@ArturBieniek4](https://github.com/ArturBieniek4) the correct way to track fixed-in is to go to the original bug ([#6945](https://github.com/verilator/verilator/issues/6945)) and edit the comment [...]

Understood, I did that now in all issues I closed as completed.

wsnyder, 2026-09-02:
Thanks, BTW I should have started my last comment with "thanks for cleaning up!".  Thanks for cleaning up! ;)