verilator/verilator #3808: Signal in an active generate block causes compiler error `Can't find definition of variable ...`

goekce, 2022-12-10:
```
module m;
  if (1) logic a;

  initial
    $display(a);
endmodule
```

Can be found in LRM page 796.
```
$ verilator --binary m.sv
%Error: main.sv:5:12: Can't find definition of variable: 'a'
    5 |   $display(a);
```

Version: Verilator 5.003 devel rev v5.002-137-ga9ff0a0f3. (latest) on Archlinux

Can someone confirm this?

wsnyder, 2022-12-11:
The code on page 796 is showing how naming works, it is not the case you indicated, it doesn't have the display which is the critical part. IEEE says that the if (1) gets a hierarchical name, so I do not believe your code is valid.  You need to use a hierarchical name (as described in IEEE) to do the display, or simply put the initial under the if statement.

```
module m;
  if (1) begin
    logic a;
    initial $display(a);
end
endmodule

```

closed 2022-12-11 by wsnyder

goekce, 2022-12-11:
You are right.

> You need to use a hierarchical name

For example:
```sv
module m;
  if (1) logic a;

  initial
    $display(genblk1.a);
endmodule
```
Thanks Wilson!