verilator/verilator #3809: parameterized type ignores parameter

jeras, 2022-12-10:
I modified `t/t_class_param.v` to showcase this issue.

The first issue is, that the parameter for a parameterized type is ignored, this part of the code properly fails.
I wrapped some code with `/* verilator lint_off WIDTH */` to avoid a build time error,
otherwise the example should pass without it.

There is also the commented out code, where class specialization is done separately from the the type usage, this ends in a parsing error.

I am using similar code with Vivado 2022.2 simulator, where it works properly.

jeras, 2022-12-10:
this is the code:
https://github.com/verilator/verilator/pull/3810/commits/5465a79fb0714c92ac6d1bffbbee9734a8db2288

wsnyder, 2022-12-11:
Confirmed, thanks for the test (#3810)

Seems the new class parameter support (#3541) didn't yet support typedefs, it looks like internally LinkDot is resolving the typedef immediately rather than waiting for parameters to resolve.  Might also be wrong before that point.

@kozdra maybe you'll get a chance to look into this, otherwise needs debug.

solomatnikov, 2023-06-07:
This test passed at c2483189580e0c0b9ba87a9087d1a1032ba5c011

closed 2023-06-07 by wsnyder