verilator/verilator #3815: Bits left unset in function call

ZipCPU, 2022-12-13:
Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

I started chasing this down with an error in the statement:

    wire [839:0] large;
    wire [103:0] value;

    // ...

    assign large = { {(840-104){1'b0}}, value };

In the VCD file, bit 124 was set in `large` following this assignment.  When I then applied bit-wise operations on `large`, bits were getting set that shouldn't have been.

I then managed to chase this back further to the function two clock cycles before,

    function [W-1:0] reverse(input [W-1:0] val);
        integer i;
    begin
        for(i=0; i<W; i=i+1)
            reverse[W-1-i] = val[i];    
    end endfunction

The purpose of this function is to reverse the bits in the input `val` provided.  Key to this bug is that the function never initializes all of `reverse`.  It shouldn't need to, since all of the 104 bits produced by the function are initialized internally within the function.  However, because Verilator maps large vectors to 32-bit words in C/C++, what happens is that the upper (128-104=) 24 bits are not getting initialized.  These bits then contribute to the following logic in ... bad ways.  Specifically, I'm shifting and then ORing this value into an 840-bit gearbox, and some of the other (840-104) bits are getting set in that gearbox--thus breaking the rest of the algorithm.

For now I'm testing a fix whereby I set `reverse=0` prior to the for loop as a work around.

_What 'verilator --version' are you using?  Did you try it with the git master version?_

I tested with `Verilator 5.003 devel rev v5.002-41-g04f68cbd9`

_What OS and distribution are you using?_

Linux Ubuntu

_May we assist you in trying to fix this in Verilator yourself?_

<Gasp!>  Frankly, I wouldn't know where to start.

ZipCPU, 2022-12-14:
> Please edit this to show the issue

Thank you for your encouragement.  Please see the attached Verilog file (in spite of its _v.txt suffix)

[vlater_v.txt](https://github.com/verilator/verilator/files/10223916/vlater_v.txt)

This file produces the output found in the attached vlater.txt file when using Verilator.

[vlater.txt](https://github.com/verilator/verilator/files/10223923/vlater.txt)

Now for an explanation: I'm still using your CRC function, but now I'm bit-wise oring the reversed bits with a 128-bit accumulator.  The reversed bits are defined as 24-bits of zeros, followed by the output of this reverse.  When you OR 128bit numbers together, for which only the bottom 104 bits are potentially non-zero bits, the top 24-bits _should_ remain zero.  In this case, only the top 17-bits are remaining zero--not the top 24.

If you look through the various loops, you'll see that the top 24-bits of result, the 24-bits which should be non-zero, are (somehow) set to 24'h0000_7F !  This change happens within the first clock or two, and then remains for the rest of the simulation.

I've tested this Verilog on both Verilator and Icarus.  Under Icarus it does what I would consider to be the "right thing", whereas under Verilog it has the problem shown in vlater.txt.

If I, on the other hand, add the line `reverse = 0;` to the beginning of the function just before the `for` loop, then everything works as expected.  This leads me to believe that this is a "failure to initialize unused bits" problem.

Dan

ZipCPU, 2022-12-14:
Also for reference, here's a Makefile and a C++ wrapper you can use to test this.

[vlater_mk.txt](https://github.com/verilator/verilator/files/10224017/vlater_mk.txt)
[vlater_cpp.txt](https://github.com/verilator/verilator/files/10224018/vlater_cpp.txt)

Dan

wsnyder, 2022-12-15:
Thanks for the good example, a C++ variable is uninitialized, I need to consider how to fix it without slowing everything down.

closed 2022-12-23 by wsnyder

ZipCPU, 2022-12-24:
Thank you!  I'm looking forward to testing this fix out.