verilator/verilator #3817: Heap use-after-free introduced in "Dfg: resolve multi-driven signal ranges #3743"

jevinskie, 2022-12-16:
After #3743, litex_sim was crashing with a segfault. I've reduced the issue to the following minimal repro (thanks to https://github.com/renatahodovan/picireny for reducing from 7k+ lines) and found it is a use-after-free after running under ASan.

```verilog
module sim ();
  wire [2:0] c_b_a;
  wire       a;
  wire       b;
  wire       c;
  assign c_b_a = {c, {b, a}};
endmodule
```

I'll try and see if I can fix this myself and add a regression test.

Version: `Verilator 5.005 devel rev v5.004-9-g7a1545751`
OS: MacOS 13.0 arm64

ASan backtrace:
```
==85399==ERROR: AddressSanitizer: heap-use-after-free on address 0x000107303058 at pc 0x00010017047c bp 0x00016fdfb880 sp 0x00016fdfb878
READ of size 8 at 0x000107303058 thread T0
    #0 0x100170478 in DfgVertex::dtypep() const V3Graph.h:331
    #1 0x1002d49f8 in DfgVertex::width() const V3Dfg.h:400
    #2 0x100323ac4 in AstToDfgVisitor::addDriver(FileLine*, unsigned int, DfgVertex*, std::__1::vector<AstToDfgVisitor::Driver, std::__1::allocator<AstToDfgVisitor::Driver> >&) const V3DfgAstToDfg.cpp:282
    #3 0x1003237f0 in AstToDfgVisitor::canonicalizePacked()::'lambda'(DfgEdge&, unsigned long)::operator()(DfgEdge&, unsigned long) const V3DfgAstToDfg.cpp:305
    #4 0x100323694 in decltype(static_cast<AstToDfgVisitor::canonicalizePacked()::'lambda'(DfgEdge&, unsigned long)&>(fp)(static_cast<DfgEdge&>(fp0), static_cast<unsigned long>(fp0))) std::__1::__invoke<AstToDfgVisitor::canonicalizePacked()::'lambda'(DfgEdge&, unsigned long)&, DfgEdge&, unsigned long>(AstToDfgVisitor::canonicalizePacked()::'lambda'(DfgEdge&, unsigned long)&, DfgEdge&, unsigned long&&) type_traits:3918
    #5 0x100323650 in void std::__1::__invoke_void_return_wrapper<void, true>::__call<AstToDfgVisitor::canonicalizePacked()::'lambda'(DfgEdge&, unsigned long)&, DfgEdge&, unsigned long>(AstToDfgVisitor::canonicalizePacked()::'lambda'(DfgEdge&, unsigned long)&, DfgEdge&, unsigned long&&) invoke.h:61
    #6 0x100323638 in std::__1::__function::__alloc_func<AstToDfgVisitor::canonicalizePacked()::'lambda'(DfgEdge&, unsigned long), std::__1::allocator<AstToDfgVisitor::canonicalizePacked()::'lambda'(DfgEdge&, unsigned long)>, void (DfgEdge&, unsigned long)>::operator()(DfgEdge&, unsigned long&&) function.h:178
    #7 0x1003224dc in std::__1::__function::__func<AstToDfgVisitor::canonicalizePacked()::'lambda'(DfgEdge&, unsigned long), std::__1::allocator<AstToDfgVisitor::canonicalizePacked()::'lambda'(DfgEdge&, unsigned long)>, void (DfgEdge&, unsigned long)>::operator()(DfgEdge&, unsigned long&&) function.h:352
    #8 0x1002e783c in std::__1::__function::__value_func<void (DfgEdge&, unsigned long)>::operator()(DfgEdge&, unsigned long&&) const function.h:505
    #9 0x1002e7760 in std::__1::function<void (DfgEdge&, unsigned long)>::operator()(DfgEdge&, unsigned long) const function.h:1182
    #10 0x1002d5150 in DfgVertex::forEachSourceEdge(std::__1::function<void (DfgEdge&, unsigned long)>) V3Dfg.h:670
    #11 0x1002f6994 in AstToDfgVisitor::canonicalizePacked() V3DfgAstToDfg.cpp:302
    #12 0x1002f62a8 in AstToDfgVisitor::AstToDfgVisitor(AstModule&, V3DfgOptimizationContext&) V3DfgAstToDfg.cpp:593
    #13 0x1002f60a0 in AstToDfgVisitor::AstToDfgVisitor(AstModule&, V3DfgOptimizationContext&) V3DfgAstToDfg.cpp:587
    #14 0x1002f5fd8 in AstToDfgVisitor::apply(AstModule&, V3DfgOptimizationContext&) V3DfgAstToDfg.cpp:599
    #15 0x1002f5ef0 in V3DfgPasses::astToDfg(AstModule&, V3DfgOptimizationContext&) V3DfgAstToDfg.cpp:604
    #16 0x1003ef5ec in V3DfgOptimizer::optimize(AstNetlist*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) V3DfgOptimizer.cpp:269
    #17 0x100008c14 in process() Verilator.cpp:246
    #18 0x100001c8c in verilate(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) Verilator.cpp:629
    #19 0x10000110c in main Verilator.cpp:754
    #20 0x186d23e4c  (<unknown module>)

0x000107303058 is located 40 bytes inside of 136-byte region [0x000107303030,0x0001073030b8)
freed by thread T0 here:
    #0 0x1047422e0 in wrap__ZdlPv+0x74 (libclang_rt.asan_osx_dynamic.dylib:arm64e+0x4e2e0)
    #1 0x1002d9ea8 in DfgConcat::~DfgConcat() V3DfgVertices.h:155
    #2 0x1002d4fe0 in DfgVertex::unlinkDelete(DfgGraph&) V3Dfg.cpp:422
    #3 0x100323af8 in AstToDfgVisitor::addDriver(FileLine*, unsigned int, DfgVertex*, std::__1::vector<AstToDfgVisitor::Driver, std::__1::allocator<AstToDfgVisitor::Driver> >&) const V3DfgAstToDfg.cpp:283
    #4 0x100323a9c in AstToDfgVisitor::addDriver(FileLine*, unsigned int, DfgVertex*, std::__1::vector<AstToDfgVisitor::Driver, std::__1::allocator<AstToDfgVisitor::Driver> >&) const V3DfgAstToDfg.cpp:280
    #5 0x1003237f0 in AstToDfgVisitor::canonicalizePacked()::'lambda'(DfgEdge&, unsigned long)::operator()(DfgEdge&, unsigned long) const V3DfgAstToDfg.cpp:305
    #6 0x100323694 in decltype(static_cast<AstToDfgVisitor::canonicalizePacked()::'lambda'(DfgEdge&, unsigned long)&>(fp)(static_cast<DfgEdge&>(fp0), static_cast<unsigned long>(fp0))) std::__1::__invoke<AstToDfgVisitor::canonicalizePacked()::'lambda'(DfgEdge&, unsigned long)&, DfgEdge&, unsigned long>(AstToDfgVisitor::canonicalizePacked()::'lambda'(DfgEdge&, unsigned long)&, DfgEdge&, unsigned long&&) type_traits:3918
    #7 0x100323650 in void std::__1::__invoke_void_return_wrapper<void, true>::__call<AstToDfgVisitor::canonicalizePacked()::'lambda'(DfgEdge&, unsigned long)&, DfgEdge&, unsigned long>(AstToDfgVisitor::canonicalizePacked()::'lambda'(DfgEdge&, unsigned long)&, DfgEdge&, unsigned long&&) invoke.h:61
    #8 0x100323638 in std::__1::__function::__alloc_func<AstToDfgVisitor::canonicalizePacked()::'lambda'(DfgEdge&, unsigned long), std::__1::allocator<AstToDfgVisitor::canonicalizePacked()::'lambda'(DfgEdge&, unsigned long)>, void (DfgEdge&, unsigned long)>::operator()(DfgEdge&, unsigned long&&) function.h:178
    #9 0x1003224dc in std::__1::__function::__func<AstToDfgVisitor::canonicalizePacked()::'lambda'(DfgEdge&, unsigned long), std::__1::allocator<AstToDfgVisitor::canonicalizePacked()::'lambda'(DfgEdge&, unsigned long)>, void (DfgEdge&, unsigned long)>::operator()(DfgEdge&, unsigned long&&) function.h:352
    #10 0x1002e783c in std::__1::__function::__value_func<void (DfgEdge&, unsigned long)>::operator()(DfgEdge&, unsigned long&&) const function.h:505
    #11 0x1002e7760 in std::__1::function<void (DfgEdge&, unsigned long)>::operator()(DfgEdge&, unsigned long) const function.h:1182
    #12 0x1002d5150 in DfgVertex::forEachSourceEdge(std::__1::function<void (DfgEdge&, unsigned long)>) V3Dfg.h:670
    #13 0x1002f6994 in AstToDfgVisitor::canonicalizePacked() V3DfgAstToDfg.cpp:302
    #14 0x1002f62a8 in AstToDfgVisitor::AstToDfgVisitor(AstModule&, V3DfgOptimizationContext&) V3DfgAstToDfg.cpp:593
    #15 0x1002f60a0 in AstToDfgVisitor::AstToDfgVisitor(AstModule&, V3DfgOptimizationContext&) V3DfgAstToDfg.cpp:587
    #16 0x1002f5fd8 in AstToDfgVisitor::apply(AstModule&, V3DfgOptimizationContext&) V3DfgAstToDfg.cpp:599
    #17 0x1002f5ef0 in V3DfgPasses::astToDfg(AstModule&, V3DfgOptimizationContext&) V3DfgAstToDfg.cpp:604
    #18 0x1003ef5ec in V3DfgOptimizer::optimize(AstNetlist*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) V3DfgOptimizer.cpp:269
    #19 0x100008c14 in process() Verilator.cpp:246
    #20 0x100001c8c in verilate(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) Verilator.cpp:629
    #21 0x10000110c in main Verilator.cpp:754
    #22 0x186d23e4c  (<unknown module>)

previously allocated by thread T0 here:
[... 33 lines cut ...]
  0x007020e80620: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x007020e80630: fd fd fa fa fa fa fa fa fa fa 00 00 00 00 00 00
  0x007020e80640: 00 00 00 00 00 00 00 00 00 00 00 00 fa fa fa fa
  0x007020e80650: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==85399==ABORTING
```

gezalore, 2022-12-16:
Thanks for tracking this down (and for telling me about picireny), and for attempting a fix, I'm happy to assist if required.

wsnyder, 2022-12-16:
Can you share the command line, grammar, etc you used with picireny?  Or better, consider a pull request documenting the process in docs/internals.rst or as a separate document?

jevinskie, 2022-12-16:
> Can you share the command line, grammar, etc you used with picireny? Or better, consider a pull request documenting the process in docs/internals.rst or as a separate document?

Of course, I’ll make a separate PR for that. In the mean time, the simple clocker example in the following repo should work out if the box and I can fix up the example reduction of this case to not require any out-of-tree litex files.

https://github.com/jevinskie/delta-debug-verilog-test

jevinskie, 2022-12-18:
#3820 attempts to fix this (will update with a regression test soon). My understanding of verilator internals is almost nil so I just pulled out my trusty trick of using a work-list. @gezalore, since you know more, does this approach make any sense?

wsnyder, 2022-12-21:
Thanks for tracking this down and making your first pull.

PR #3820 (2022-12-21): AstToDfg addDriver: cache rhs width to avoid use-after-free
This is my uneducated attempt to resolve #3817. It uses a work-list so the concat nodes aren't freed while in a nested addDriver() call.

I'll update this PR with a regression test once I learn how those are done in verilator.
closed 2022-12-21 by wsnyder