verilator/verilator #3826: Use of structs and struct-type arrays in constant functions

StefanoBettelli, 2022-12-25:
In **parametric** elaboration of a circuit it is sometimes useful to resort to a function whose result is saved to an **array-type local parameter** for use in a **generate block**. The following snippet is working:
```
module mymodule;
   localparam NUM_SUBELEMENTS = <parametrically determined>;

   typedef integer list_t[NUM_SUB_ELEMENTS]; // integer-type array
   
   function list_t make_list;
      for (int k=0; k<$size(make_list); ++j) begin
         make_list[j] = <your algorithm here>;
      end
   endfunction

   localparam list_t xx = make_list();

   for (genvar j=0; j<$size(xx); +++j)
        <connect wires according to const values in xx[j] >
endmodule
```
But if the base type of the array is not a simple variable, then verilator fails to elaborate:
```
   typedef struct {
      integer 	  a;
      integer 	  b;
   } mystruct;

   typedef mystruct list_t[NUM_SUBELEMENTS]; // stuct-type array
   
   function list_t make_list;
      for (int k=0; k<$size(make_list); ++j) begin
         make_list[j].a = <your algorithm here>;
         make_list[j].b = <your algorithm here>;
      end
   endfunction
```
gives
```
Verilating design XXX.sv
%Error: XXX.sv:166:27: Expecting expression to be constant, but can't determine constant for FUNCREF 'make_list'
                                  : ... In instance XXX
        XXX.sv:160:17: ... Location of non-constant ASSIGN: LHS isn't simple variable
        XXX.sv:166:27: ... Called from make_list() with parameters:
  166 |    localparam list_t xx = make_list();
      |                           ^~~~~~~~~

```
The request is of course whether it is possible to support this case.

wsnyder, 2022-12-25:
This is likely a side effect of the new unpacked structure support.  Perhaps you could look at making a pull to support this?  Likely V3Simulate is what needs fixing.

StefanoBettelli, 2022-12-26:
I rerun the same verilation with master commit c3b4bc8 (the one before the commit with title "Support unpacked structs (#3802)"), but the result was not different, so, maybe it is not due to structure support.

I may have a look in the future to how the AST is handled in verilator, but it will certainly take a lot of time. Moreover, it would be nearly useless if issue #3829 is not solved first.

closed 2026-10-18 by wsnyder