verilator/verilator #3829: Very large memory footprint while handling constant arrays

StefanoBettelli, 2022-12-26:
Not really a bug, but it makes parametric circuit generation almost impossible.
The following code initialises an array of ~5k integer elements in a function evaluated in constant context (the mylist local parameter would eventually be used in a generate block to create a custom circuit during elaboration, with a more complicated algorithm, of course).
On my laptop even just linting it ("verilator --lint-only mytest.sv") uses nearly **6GB in memory**, let alone the 64bit case.
```
module mytest #(parameter NBITS = 32) ();
   localparam NUMCIRCUITS = 5*NBITS*NBITS;
   typedef integer unsigned list_t[NUMCIRCUITS];

   function list_t make_list;
      for (int j=0; j<NUMCIRCUITS; ++j) make_list[j] = j;
   endfunction

   localparam list_t mylist = make_list();
   initial
     $display($size(mylist), mylist[0], mylist[NUMCIRCUITS-1]);
endmodule
```
```
builds/mytest/simulator +verilator+rand+reset+2
       5120         0      5119
```

What 'verilator --version' are you using?  Did you try it with the git master version?

**5.005 devel rev v5.004-28-ga86ded97c**

What OS and distribution are you using?

**Linux Ubuntu 22.10**

wsnyder, 2022-12-29:
Two problems, the permanent leak and the peak usage.

For leaks, In V3Simulate the AstInitArray gets cloneTree'ed at each step, then pushed for cleanup at end.  The other constants under the cloneTree are never freed.  Also the default value for AstInitArray is never freed.

For peak usage, the AstInitArrays need to be reclaimed immediately when reassigned.

wsnyder, 2025-09-28:
Example uses only 35MB with current master, and confirmed no leak using --enable-asan.  There were improvements in the algorithm earlier.

closed 2025-09-28 by wsnyder