verilator/verilator #3831: Incorrect/misleading error message for Warning-WIDTHCONCAT

AndrewNolte, 2022-12-27:
Tested on 4b6e15d0eb3ab7f4c1860dc7ba80016b9c4aafa9 (Dec 6), but should also be on up to date master.

```
%Warning-WIDTHCONCAT: t/t_concat_param.v:22:57: Unsized numbers/parameters not allowed in concatenations.
                                              : ... In instance t
   22 |     localparam logic [total_len-1:0] offsets_i = {value1, arr1[len1-1:0]};
      |                                                         ^
```

```
    localparam logic            value1    = '0;
```

However, this is coming from a localparam that has a lhs width of 1, so I don't think this error should be here. I think maybe that lhs width is being optimized out, and so we get the error here. If value1 is changed to 0 instead of '0, it passes. Test case attached.

wsnyder, 2022-12-27:
This is as intended, while the size by IEEE will default to 32 bits, this is not something code should rely on, which is the intent if the warning.  Use "parameter int" or 32'b0 instead.

AndrewNolte, 2023-01-20:
I totally agree! But I think in that case the error should point at ```    localparam logic            value1    = '0;```, not line 22.

closed 2023-01-20 by wsnyder