verilator/verilator #3836: Inconsistent naming of generate scope arrays

AndrewNolte, 2022-12-30:
Sometimes the generated scope will output __BRA__0__KET__, making some things that were previously vpi accessible obscured by these names that should be internal to Verilator. These should be t.arr[0], t.subs[0], t.subs[0].subsub, etc.
Full output shown here https://github.com/verilator/verilator/pull/3835. 

```
    SCOPE 0x55da6510d918: t.arr__BRA__1__KET__
    SCOPE 0x55da6510d988: t.arr__BRA__2__KET__
    SCOPE 0x55da6510d9f8: t.arr__BRA__3__KET__
    SCOPE 0x55da6510da68: t.arr__BRA__4__KET__
    SCOPE 0x55da6510dad8: t.arr__BRA__5__KET__
    SCOPE 0x55da6510db48: t.arr__BRA__6__KET__
    SCOPE 0x55da6510dbb8: t.sub
       VAR 0x55da65111208: subsig1
       VAR 0x55da65111288: subsig2
    SCOPE 0x55da6510dbf0: t.subs__BRA__1__KET__
    SCOPE 0x55da6510dc28: t.subs__BRA__1__KET__.subsub
       VAR 0x55da65111348: subsig1
       VAR 0x55da651113c8: subsig2
    SCOPE 0x55da6510dc60: t.subs__BRA__2__KET__
    SCOPE 0x55da6510dc98: t.subs__BRA__2__KET__.subsub
       VAR 0x55da65111488: subsig1
       VAR 0x55da65111508: subsig2
    SCOPE 0x55da6510dcd0: t.subs__BRA__3__KET__
    SCOPE 0x55da6510dd08: t.subs__BRA__3__KET__.subsub
       VAR 0x55da651115c8: subsig1
       VAR 0x55da65111648: subsig2
    SCOPE 0x55da6510dd40: t.subs__BRA__4__KET__
    SCOPE 0x55da6510dd78: t.subs__BRA__4__KET__.subsub
       VAR 0x55da65111708: subsig1
       VAR 0x55da65111788: subsig2
    SCOPE 0x55da6510ddb0: t.subs__BRA__5__KET__
    SCOPE 0x55da6510dde8: t.subs__BRA__5__KET__.subsub
       VAR 0x55da65111848: subsig1
       VAR 0x55da651118c8: subsig2
    SCOPE 0x55da6510de20: t.subs__BRA__6__KET__
    SCOPE 0x55da6510de58: t.subs__BRA__6__KET__.subsub
```

wsnyder, 2022-12-31:
Thanks for the test case in #3835.  There is some translation between the VPI routines and cell names, so ideally the test would be using VPI accessors to show the misnaming breaks the VPI accessing - be sure to run the test against another simulator first to make sure the arrayed cells are IEEE-expected.

Perhaps you could then improve the pull request to find a fix?

AndrewNolte, 2023-01-02:
Yeah, I can definitely make a better test, I thought maybe it was a quick fix. Do you know where I should look to get started on a fix?

AndrewNolte, 2023-01-02:
I was able to fix it by modifying  void visit(AstCellInline* nodep) in v3emitcsyms, to do the pretty name instead of dedot. I think I'll change the AstScope* case as well, although I didn't run into issues with that one in my test.

AndrewNolte, 2023-01-02:
I have one more case of __BRA__ __KET__ sneaking through, I'm going to try and solve that before merging

AndrewNolte, 2023-01-03:
Actually it doesn't need any more changes, it should be fine to merge!

wsnyder, 2023-01-03:
I hadn't noticed that your pull request for some reason didn't run all the regression tests, and when merged to master it failed, so I reverted it.  Maybe you didn't allow edits by third parties?  Anyhow please fix all tests and make a new pull request, thanks.

AndrewNolte, 2023-01-05:
Yeah vpi_module had issues with escaped character module names, apparently those need to be not be pretty printed in order for the access to work. It passes everything now on the latest build after I fixed that. https://github.com/AndrewNolte/verilator/actions/runs/3842328026

closed 2023-02-02 by AndrewNolte