verilator/verilator #3837: Multi-bit signals in VCD/FST files are suffixed with <space>[msb:lsb]

pieter3d, 2022-12-31:
This seems to be a recent development. In both VCD and FST files, the signal names (when wider than 1) are now suffixed by a space and `[msb:lsb]`. The space actually causes the VCD files to be illegally formatted, as the expected format is `$var [ var_type size identifier_code reference ] $end`, but now `reference` looks like e.g. `signal_name [31:0]`

GTKWave appears to be immune to this somehow.

---

What 'verilator --version' are you using?  Did you try it with the git master version?
git master version

What OS and distribution are you using?
Ubuntu 22 LTS

wsnyder, 2022-12-31:
This has not changed. The standard allows spaces between tokens, and traces I have from 1990's typically have spaces.  Whatever tool is not handling this is broken.

closed 2022-12-31 by wsnyder

pieter3d, 2022-12-31:
Hmm ok. I definitely have seen behavior changes after moving to verilator git head, but perhaps I've overlooked something else.
The tool where I saw this was my own, [simview](https://github.com/pieter3d/simview).