verilator/verilator #3843: Debian maintainer

tarikgraba, 2023-01-05:
Hi,

The Debian package has been removed form the next-stable Debian release.

https://tracker.debian.org/news/1376559/verilator-removed-from-testing/

Which means that at some point it will also be removed from Ubuntu official repos and one of the installation methods described in the documentation will not  work anymore.

_The removal is related to the maintenance of the SystemC package which is a dependence of the verilator package and has also been removed_

If somebody feels to help resolve thees issues, it would be of great help for the community.

wsnyder, 2023-01-05:
@aelmahmoudy is this something you might be able to look at?  Otherwise I can contact Accellera to see if they can look into it.

Also I noted that the Verilator debian files aren't pointing to the stable branch here, which they should be. They are stuck on a very old tarball which we don't make any more.

tarikgraba, 2023-01-06:
I think @tijuca is the now the active maintainer and the problem seems to be already resolved (at least for SystemC).

https://tracker.debian.org/news/1405606/accepted-systemc-234-1-source-into-unstable/

Thanks to @tijuca and @aelmahmoudy for their time/work.

tijuca, 2023-01-06:
Hi,

FTR, the de-facto only maintainer for systemc in Debian until now was @aelmahmoudy who is taking some time off from Debian so far I know and I'm aware.

An Debian user was reaching the [mailing list](https://alioth-lists.debian.net/pipermail/pkg-electronics-devel/2023-January/009594.html) of the Debian Electronics Packaging group and was telling about his fear systemc/verilator wont be shipped in bookworm.
As I've take some action on systemc in the past, and I already has started to look at the current reason behind the removal in Debian testing I offered to have a look again and fix outstanding build issues so systemc can be part of testing again. This does not necessary means I will and can do this in future times, it would be nice and helpful if further people would start to take over maintainer ship for e.g systemc!

The user who has contacted the ML is willing to give it a try for taking over some package maintenance for verilator. I myself don't use systemc nor verilator on my own, so it's difficult to do some technical work of packaging stuff but don't do any testing before uploading stuff to the Debian archive, and I wont do something like that without at least some testing by experienced users.

wsnyder, 2023-01-06:
Great!  This page https://tracker.debian.org/pkg/verilator though still shows Verilator as removed.  Also the metadata for Verilator needs to be updated to use the stable repo here to pull a newer version.

Ideally Verilator and systemc would be built with C++20, but only if that's the default compiler on the proposed distro.

tarikgraba, 2023-01-13:
@wsnyder the package is back into testing. I think sombedy should help @tijuca  to update the package. 

The  Debian package still tracks te old repo:

https://salsa.debian.org/electronics-team/verilator/-/blob/master/debian/upstream/metadata

wsnyder, 2023-01-13:
The repository settings should be as below.  It's important to use the "stable" tag, but I don't know how to tell Debian that, and also I don't know if other steps are needed to have debian pull it from git.  Presumably there's many other packages' examples to copy from.

```
Repository: https://github.com/verilator/verilator.git
Repository-Browse: https://github.com/verilator/verilator
```

tijuca, 2023-01-13:
Investigate and updating these data is a usual step while doing package updates, so no worries, this is the really easy part.

Doing a quick and dirty import of the most recent upstream source, that was tagged version 5.004, some days ago I did see problems while the package  configuration step. I haven't looked at this closer and probably wont have to to do it as my focus is definitely on other packages right now. If possible I can add the example output of the configure run later.

Debian is preparing GCC-12 and LLVM 14 for the bookworm release.

wsnyder, 2023-01-13:
If you want, give me a recipie to get the messages using a VM, suitable for someone that has never done it before, I'll look at the messages and clean up, making a new Verilator release if needed. Thanks.

What is the date this needs to be done to hit the next release?

tijuca, 2023-01-15:
I got a bit of time yesterday and could look deeper at the packaging of verilator in Debian.

Looks good for now, I can build a new version of 5.004 as Debian packages. But I can't really test the current state as I'm not in the universe of verilator and the Verilog simulator and lint system. Comparing the old package with the new one I think it will work as before.

Lintian is showing some small issues or suggested improvements, nothing dramatic or blocking.

```
W: verilator: no-manual-page [usr/bin/verilator_gantt]
W: verilator: no-manual-page [usr/bin/verilator_profcfunc]
W: verilator: privacy-breach-generic [<script async src="https://www.googletagmanager.com/gtag/js?id=g-d27b0c9qeb">] (https://www.googletagmanager.com/gtag/js?id=g-d27b0c9qeb) [usr/share/doc/verilator/html/changes.html]
# Note: can be ignored here, comes from sphinx package that fills in that content.
...
I: verilator: spelling-error-in-binary Savable Saveable [usr/bin/verilator_bin]
I: verilator: spelling-error-in-binary Savable Saveable [usr/bin/verilator_bin_dbg]
I: verilator: spelling-error-in-binary couL could [usr/bin/verilator_bin]
```

I'm not sure about the mentioned spelling issues, that's probably a false positive. Getting some man pages for the two mentioned binaries would be nice.

The package preparation can be visit here: https://salsa.debian.org/tijuca/verilator

I've pushed packages to https://people.debian.org/~tijuca/verilator/

A remaining issue is the failure happen if `make test` is called, for now I disabled this target in the build as I wasn't able to solve the problem in time. Looks to me like a linking issue for now. If possible I'd like to get some help on this.

```
...
	make -C obj_vlt/t_a2_first_sc -f /build/verilator-5.004/test_regress/Makefile_obj --no-print-directory VM_PREFIX=Vt_a2_first_sc TEST_OBJ_DIR=obj_vlt/t_a2_first_sc CPPFLAGS_DRIVER=-DT_A2_FIRST_SC  OPT_FAST=-O0 OPT_GLOBAL=-O0 Vt_a2_first_sc    > obj_vlt/t_a2_first_sc/vlt_gcc.log
driver: Entering directory '/build/verilator-5.004/test_regress/obj_vlt/t_a2_first_sc'
ccache g++ -g -O2 -ffile-prefix-map=/build/verilator-5.004=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -I.  -MMD -I/build/verilator-5.004/test_regress/../include -I/build/verilator-5.004/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=1 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow        -std=gnu++14 -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a2_first_sc -DVM_PREFIX=Vt_a2_first_sc -DVM_PREFIX_INCLUDE="<Vt_a2_first_sc.h>" -DVM_PREFIX_ROOT_INCLUDE="<Vt_a2_first_sc___024root.h>" -DT_A2_FIRST_SC   -O0 -c -o Vt_a2_first_sc__main.o ../../obj_vlt/t_a2_first_sc/Vt_a2_first_sc__main.cpp
ccache g++ -g -O2 -ffile-prefix-map=/build/verilator-5.004=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -I.  -MMD -I/build/verilator-5.004/test_regress/../include -I/build/verilator-5.004/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=1 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow        -std=gnu++14 -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a2_first_sc -DVM_PREFIX=Vt_a2_first_sc -DVM_PREFIX_INCLUDE="<Vt_a2_first_sc.h>" -DVM_PREFIX_ROOT_INCLUDE="<Vt_a2_first_sc___024root.h>" -DT_A2_FIRST_SC   -O0 -c -o verilated.o /build/verilator-5.004/test_regress/../include/verilated.cpp
ccache g++ -g -O2 -ffile-prefix-map=/build/verilator-5.004=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -I.  -MMD -I/build/verilator-5.004/test_regress/../include -I/build/verilator-5.004/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=1 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow        -std=gnu++14 -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a2_first_sc -DVM_PREFIX=Vt_a2_first_sc -DVM_PREFIX_INCLUDE="<Vt_a2_first_sc.h>" -DVM_PREFIX_ROOT_INCLUDE="<Vt_a2_first_sc___024root.h>" -DT_A2_FIRST_SC   -O0 -c -o verilated_vcd_c.o /build/verilator-5.004/test_regress/../include/verilated_vcd_c.cpp
ccache g++ -g -O2 -ffile-prefix-map=/build/verilator-5.004=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -I.  -MMD -I/build/verilator-5.004/test_regress/../include -I/build/verilator-5.004/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=1 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow        -std=gnu++14 -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a2_first_sc -DVM_PREFIX=Vt_a2_first_sc -DVM_PREFIX_INCLUDE="<Vt_a2_first_sc.h>" -DVM_PREFIX_ROOT_INCLUDE="<Vt_a2_first_sc___024root.h>" -DT_A2_FIRST_SC   -O0 -c -o verilated_threads.o /build/verilator-5.004/test_regress/../include/verilated_threads.cpp
[... cut ...]
> What is the date this needs to be done to hit the next release?

There is enough time I'd say, as long verilator is staying in testing it will be part of the next release. The next critical date is 02-12-2023 as this is the soft freeze, means by this date no new versions bumps will be typically allowed. See https://release.debian.org/testing/freeze_policy.html#soft
My guess right now is that there are no big problems left so I could upload a version 5.004 to unstable and package can migrate to testing typically 5 days later (if all depends are fulfilled).

ldoolitt, 2023-01-16:
> A remaining issue is the failure happen if make test is called, for now I disabled this target in the build as I wasn't able to solve the problem in time.

I can "fix" that problem by commenting out
```
CPPFLAGS += $(CFG_CXXFLAGS_STD_NEWEST)
```
in `include/verilated.mk.in`.  Patch (such that it is) attached.
[notnewest.txt](https://github.com/verilator/verilator/files/10423178/notnewest.txt)

Maybe relates to @wsnyder comment in issue #3847
> It's not like std=c99 as all C libraries are generally compatible, but C++ libraries are more complicated so it is better to avoid std= unless it is needed to compile.

tijuca, 2023-01-16:
Ahh, o.k. that is working, but a bit taking a big hammer.

Currently I modifying `configure.ac` (line 358) before calling `autoconf` so the output of `lsb_release` will be found as match. The options `*Arch*Linux* | *Ubuntu*22.04*` are to strict to detect a Debian system. Adding `bookworm` at least to the list would be required, but that's not very future proof and would also exclude nearly all downstream distros as they use other names. Doing a real time detection based on the compiler would be better.

ldoolitt, 2023-01-16:
> Currently I modifying configure.ac (line 358) before calling autoconf so the output of lsb_release will be found as match.

Go read issue #3847.  Your approach is what I suggested in my original post, and @wsnyder told me that was backwards.
He says the best approach is usually to _not_ add `-std=gnu++14` or whatever, since the default is most likely to work with the system C++ library.  Not sure why `CPPFLAGS += $(CFG_CXXFLAGS_STD_NEWEST)` leaked into `include/verilated.mk.in`.
Git blame points the finger at @kamilrakoczy in commit d6126c4b32.

BTW, I can recreate the `make test` fault (and confirm the fix) in a "normal" environment on git master with Debian (pre-)Bookworm.  I'll file a separate issue for that.

wsnyder, 2023-01-18:
Please try the head with (#3881) fixed.

>W: verilator: no-manual-page [usr/bin/verilator_gantt]
>W: verilator: no-manual-page [usr/bin/verilator_profcfunc]

I added installing these using help2man, you might need to add the new help2man dependency to the build requirements. 
 
>I: verilator: spelling-error-in-binary Savable Saveable [usr/bin/verilator_bin]

Savable is an acceptable spelling of saveable (dictionary.com).

>I: verilator: spelling-error-in-binary couL could [usr/bin/verilator_bin]

Seems a false positive on some non-text.

tijuca, 2023-01-22:
Hi,
with #3881 included and some additions from #3878 to the controlling of the build process on the Debian buildds these machines are now building version 5.004 for all release [critical platforms](https://buildd.debian.org/status/package.php?p=verilator). At least this version will be part of Debian (12) bookworm as I expect the current latest version in unstable will migrated to testing.

I'd say this issue here can be closed now.

closed 2023-01-22 by wsnyder