verilator/verilator #3847: C++ compiler versions

ldoolitt, 2023-01-09:
Please reconcile issue #3843 comment
 `"Ideally Verilator and systemc would be built with C++20"`
with the configure.ac (at least as of 520f6e5f) comment
`# c++17 is the newest that Verilator is regularly tested to support`
configure.ac also has no obvious mechanism to set `-std=c++20`.

Personally, I'm not a fan of  `-std=gnu++*`, and would take those attempts, like
` _MY_CXX_CHECK_SET(CFG_CXXFLAGS_STD_NEWEST,-std=gnu++17)`
out of configure.ac.  Either this code base is portable, or it uses gnu extensions. 
The evidence I see is that it is pure c++.
I just built a fresh copy on Debian Bookworm (gnu g++ 12.2.0) using `-std=c++17` and it worked great.

Getting `-std=c++17` onto the command line required hacking on configure.ac.
Please either add Debian bookworm to the good-list
```
-*Alpine*Linux* | *Arch*Linux* | *Ubuntu*22.04*)
+*Debian*bookworm* | *Alpine*Linux* | *Arch*Linux* | *Ubuntu*22.04*)
```
or make that the default.

ldoolitt, 2023-01-09:
Also, pretty sure there's a typo in configure.ac.  Please check if
```
@@ -535,7 +535,7 @@ AC_DEFUN([_MY_CXX_CHECK_CXX_VER],
 AC_MSG_CHECKING(whether $CXX supports C++11)
 _MY_CXX_CHECK_CXX_VER()
 AC_MSG_RESULT($_my_result)
-if test "$_my_result" = "no" ; then
+if test "$_my_result" = "yes" ; then
    CXXFLAGS="$CXXFLAGS $CFG_CXXFLAGS_STD_NEWEST"
    CFG_CXX_FLAGS_CMAKE="$CFG_CXX_FLAGS_CMAKE $CFG_CXXFLAGS_STD_NEWEST"
    AC_MSG_CHECKING(whether $CXX supports C++11 with $CFG_CXXFLAGS_STD_NEWEST)
```
is what was actually intended.

ldoolitt, 2023-01-09:
I got my answer about `-std=c++20`, on g++ 12.2.0 at least.  See attached trace, 
[buildlog_20.txt](https://github.com/verilator/verilator/files/10371291/buildlog_20.txt)
that goes OK for a while but errors out building V3Dfg.cpp:
```
g++ -std=c++20 -Og -ggdb -gz -DVL_DEBUG -D_GLIBCXX_DEBUG -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/home/larry/share/verilator\" -c ../V3Dfg.cpp -o V3Dfg.o
In file included from ../V3Dfg.cpp:20:
../V3Dfg.h:745:39: error: expected ‘)’ before ‘&’ token
```
etc.

wsnyder, 2023-01-10:
The CI here uses Ubuntu 22.04 LTS (or older) which is gcc 11.3 which still defaults to C++17.

As mentioned in the comments the "gnu" extension is needed only due to Cygwin breaking otherwise.

The "no" test is intentional.  This is to default to not using any --std flags, as that is what's best to avoid problems with libstdc. Only if we can't get a working C++11 (the if "no" code) do we then try the -std flags, as older gcc defaulted to not support >= C++11 unless "asked".

wsnyder, 2023-01-10:
More specifically on 22.04 LTS

```
$ clang++ -E -dM -c clang_predefines.cpp | grep __cplusplus
#define __cplusplus 201402L
$ g++ -E -dM -c clang_predefines.cpp | grep __cplusplus
#define __cplusplus 201703L
```

wsnyder, 2023-01-10:
BTW I think I confused things above, it isn't we need a C++20 compiler, it is we want ideally a C++20 compatible std library with coroutines.

ldoolitt, 2023-01-10:
Thanks for the clarifications!
On Debian Bullseye g++ 10.2.1 I get
```
$ for n in $(seq 11 21); do echo $n; g++ -std=c++$n -E -dM -c gcc_predefines.cpp 2>/dev/null | grep __cplusplus; done
11
#define __cplusplus 201103L
12
13
14
#define __cplusplus 201402L
15
16
17
#define __cplusplus 201703L
18
19
20
#define __cplusplus 201709L
21
```
On Debian (pre-)Bookworm g++ 12.2.0 I get
```
$ for n in $(seq 11 21); do echo $n; g++ -std=c++$n -E -dM -c gcc_predefines.cpp 2>/dev/null | grep __cplusplus; done
11
#define __cplusplus 201103L
12
13
14
#define __cplusplus 201402L
15
16
17
#define __cplusplus 201703L
18
19
20
#define __cplusplus 202002L
21
```
Mildly interesting.

> This is to default to not using any --std flags, as that is what's best to avoid problems with libstdc.

I'm really more of a c programmer than a c++ programmer.  I'm used to gcc needing the `-std=c99` flag (or similar) to keep it from accepting all kinds of crazy non-standard input.  Is the situation different with c++?

> we want ideally a C++20 compatible std library with coroutines.

Is that the HAVE_COROUTINES test?  Comes out
```
checking whether g++ accepts -fcoroutines-ts... no
checking whether coroutines are supported by g++... yes
```
on Debian (pre-)Bookworm, and therefore `#define HAVE_COROUTINES 1` in src/config_build.h.

wsnyder, 2023-01-10:
>checking whether coroutines are supported by g++... yes

Then I think you are all set.

ldoolitt, 2023-01-10:
Good discussion, thanks!

I didn't mention above, but probably should have: Debian (pre-)Bookworm gcc 12.2.0 default is
```
#define __cplusplus 201703L
```
There's a pending question
> I'm used to gcc needing the -std=c99 flag (or similar) to keep it from accepting all kinds of crazy non-standard input. Is the situation different with c++?

And, should any of this discussion get captured in the documentation, before this "issue" gets closed?
Getting documentation in places where it actually gets read is quite a trick, I know from first-hand experience.

wsnyder, 2023-01-10:
It's not like std=c99 as all C libraries are generally compatible, but C++ libraries are more complicated so it is better to avoid std= unless it is needed to compile.

I added a comment around the "no", if you think more is needed a pull request to add comments (anywhere) is always welcome!

wsnyder, 2023-01-26:
I think the other pull requests and issues mentioned here have resolved the distro issues.

closed 2023-01-26 by wsnyder