verilator/verilator #3848: `SIGSEGV` in `CaseVisitor::isCaseTreeFast(AstCase*)`

zarubaf, 2023-01-09:
I am getting the following segementation fault:

```
- V3Ast.cpp:1122:     Dumping obj_dir/Vtestharness_017_const.tree
- V3SplitVar.cpp:1246:splitVariable: 
- V3Inst.cpp:623:     dearrayAll: 
- V3LinkDot.cpp:3320: linkDotArrayed: 
- V3Begin.cpp:341:    debeginAll: 
- V3Ast.cpp:1122:     Dumping obj_dir/Vtestharness_022_begin.tree
- V3Tristate.cpp:1819:tristateAll: 
- V3Ast.cpp:1122:     Dumping obj_dir/Vtestharness_023_tristate.tree
- V3Unknown.cpp:520:  unknownAll: 
- V3Ast.cpp:1122:     Dumping obj_dir/Vtestharness_024_unknown.tree
- V3DfgOptimizer.cpp:237:extract: 
- V3Hasher.cpp:86:    %Warning: Hashing node as AstNode: CASEITEM 0xa4f68620 <e1751149> {hh1280an}
- V3Hasher.cpp:86:    %Warning: Hashing node as AstNode: CASEITEM 0xa4f68f80 <e1751178> {hh1281an}
- V3Hasher.cpp:86:    %Warning: Hashing node as AstNode: CASEITEM 0xa4f698e0 <e1751207> {hh1282an}
- V3Hasher.cpp:86:    %Warning: Hashing node as AstNode: CASEITEM 0xa4f6a240 <e1751236> {hh1283an}
- V3Hasher.cpp:86:    %Warning: Hashing node as AstNode: CASEITEM 0xa4f6aba0 <e1751265> {hh1284an}
- V3Hasher.cpp:86:    %Warning: Hashing node as AstNode: CASEITEM 0xa4f6b500 <e1751294> {hh1285an}
- V3Hasher.cpp:86:    %Warning: Hashing node as AstNode: CASEITEM 0xa4f6be60 <e1751306> {hh1286ah}
- V3Ast.cpp:1122:     Dumping obj_dir/Vtestharness_025_dfg-extract.tree
- V3DfgOptimizer.cpp:244:optimize: 
- V3Ast.cpp:1122:     Dumping obj_dir/Vtestharness_026_dfg-optimize.tree
- V3Inline.cpp:708:   inlineAll: 
- V3Ast.cpp:1122:     Dumping obj_dir/Vtestharness_027_inline.tree
- V3LinkDot.cpp:3320: linkDotArrayed: 
- V3DfgOptimizer.cpp:244:optimize: 
- V3Ast.cpp:1122:     Dumping obj_dir/Vtestharness_029_dfg-optimize.tree
- V3Const.cpp:3754:   constifyAll: 
-arning-suppressed: /home/projects/workspace/zarubaf/triton-debug/subip/sys_ctrl/verif/testharness.sv:906:5: Infinite loop (condition always true)...
- V3Ast.cpp:1122:     Dumping obj_dir/Vtestharness_030_const.tree
- V3Dead.cpp:521:     deadifyDTypes: 
- V3Ast.cpp:1122:     Dumping obj_dir/Vtestharness_031_deadDtypes.tree
- V3Inst.cpp:617:     instAll: 
- V3Ast.cpp:1122:     Dumping obj_dir/Vtestharness_032_inst.tree
- V3Const.cpp:3754:   constifyAll: 
- V3Ast.cpp:1122:     Dumping obj_dir/Vtestharness_033_const.tree
- V3Scope.cpp:409:    scopeAll: 
- V3Ast.cpp:1122:     Dumping obj_dir/Vtestharness_034_scope.tree
- V3LinkDot.cpp:3326: linkDotScope: 
- V3Ast.cpp:1122:     Dumping obj_dir/Vtestharness_035_linkdot.tree
- V3Class.cpp:209:    classAll: 
- V3Ast.cpp:1122:     Dumping obj_dir/Vtestharness_036_class.tree
- V3Const.cpp:3754:   constifyAll: 
- V3Ast.cpp:1122:     Dumping obj_dir/Vtestharness_037_const.tree
- V3Dead.cpp:527:     deadifyDTypesScoped: 
- V3Ast.cpp:1122:     Dumping obj_dir/Vtestharness_038_deadDtypesScoped.tree
- V3Case.cpp:565:     caseAll: 

Program received signal SIGSEGV, Segmentation fault.
0x000000000050b771 in CaseVisitor::isCaseTreeFast(AstCase*) ()
#0  0x000000000050b771 in CaseVisitor::isCaseTreeFast(AstCase*) ()
#1  0x000000000050b92d in CaseVisitor::visit(AstCase*) ()
#2  0x00000000009ca864 in AstCase::accept(VNVisitor&) ()
#3  0x000000000043f6a7 in AstNode::iterateAndNext(VNVisitor&) ()
#4  0x000000000043f74b in AstNode::iterateChildren(VNVisitor&) ()
#5  0x0000000000503e03 in CaseVisitor::visit(AstNode*) ()
#6  0x0000000000435adc in VNVisitor::visit(AstNodeStmt*) ()
#7  0x00000000004355e9 in VNVisitor::visit(AstJumpBlock*) ()
#8  0x00000000004d36ba in AstJumpBlock::accept(VNVisitor&) ()
#9  0x000000000043f6a7 in AstNode::iterateAndNext(VNVisitor&) ()
#10 0x000000000043f74b in AstNode::iterateChildren(VNVisitor&) ()
#11 0x0000000000503e03 in CaseVisitor::visit(AstNode*) ()
#12 0x0000000000435adc in VNVisitor::visit(AstNodeStmt*) ()
#13 0x00000000004355e9 in VNVisitor::visit(AstJumpBlock*) ()
#14 0x00000000004d36ba in AstJumpBlock::accept(VNVisitor&) ()
#15 0x000000000043f6a7 in AstNode::iterateAndNext(VNVisitor&) ()
#16 0x000000000043f74b in AstNode::iterateChildren(VNVisitor&) ()
#17 0x0000000000503e03 in CaseVisitor::visit(AstNode*) ()
[... 79 lines cut ...]
#97 0x00000000005036ff in V3Case::caseAll(AstNetlist*) ()
#98 0x0000000000408994 in process() ()
#99 0x00000000004096a3 in verilate(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
#100 0x000000000040c013 in main ()
A debugging session is active.

	Inferior 1 [process 47013] will be killed.

Quit anyway? (y or n) [answered Y; input not from terminal]
%Error: export VERILATOR_ROOT=/home/projects/verilator/v5.004
%Error: ulimit -s unlimited 2>/dev/null; exec gdb /home/projects/verilator/v5.004/bin/verilator_bin_dbg --batch --quiet --return-child-result -ex "run --debug --gdbbt --no-timing -cc --build -j 8 --top-module testharness --unroll-count 2048 -Wno-UNOPTFLAT -Wno-fatal -f scripts/compile.verilator.sh" -ex 'set width 0' -ex 'bt' -ex 'quit'
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec gdb /home/projects/verilator/v5.004/bin/verilator_bin_dbg --batch --quiet --return-child-result -ex "run --debug --gdbbt --no-timing -cc --build -j 8 --top-module testharness --unroll-count 2048 -Wno-UNOPTFLAT -Wno-fatal -f scripts/compile.verilator.sh" -ex 'set width 0' -ex 'bt' -ex 'quit'
make: *** [verilate] Error 255
```

Present in the current top-of-tree version as well as the released stable version:

`Verilator 5.004 2022-12-14 rev v5.004` in the latest `4.x` version this does not occur.

What OS and distribution are you using?

CentOS 7

May we assist you in trying to fix this in Verilator yourself?
Sure 👍

wsnyder, 2023-01-10:
Thanks for reporting and looking into this.

It's strange the backtrace didn't print the argument values like it normally does.  Use --gdb and "up" until you see a call which says what the nodep is that is breaking. Then look for that pointer value in the highest number .tree file.  That will tell you the filename/linenumber that is breaking (see docs/internals.rst for how to read the dumps).

Alternatively/in addition, try to simplify the input verilog code to the smallest example that recreates the problem.

From the source alone it might be obvious what's causing the problem.  Anyhow the next step is to create a test_regress test showing the problem, then work on a pull request that makes a fix along with the now passing test..

wsnyder, 2023-02-07:
Think this was fixed with the commit above, if not please post a failing test case.

closed 2023-02-07 by wsnyder