verilator/verilator #3849: Clarification on --public-flat-rw warnings

sarutob1, 2023-01-09:
---

Original Redmine Comment
Author Name: **Wilson Snyder** (@wsnyder)
Original Date: 2019-09-19T22:52:47Z

---

I really don't like expanding the userfullness use of public.  There's two problems:

1. Users will probably set it then not ever tune more finely (how many designs have +acc+rw forever) and get bad performance forever.

2. Making some things public such as signals used to generate clocks can cause incorrect behavior.

I do understand however that users might not know what variables to r/w.  

Another counter argument maybe that some people are using --public now and this is better from a performance perspective.

At a minimum the docs need to better discourage this.

Is there any other option open?

BTW if you can please squash so you make one patch file it would be helpful.  But 0005 seems separate and was applied.

_Originally posted by @veripoolbot in https://github.com/verilator/verilator/issues/1511#issuecomment-568287784_
      
In the above post, @wsnyder, you mentioned that making signals used to generate clocks public can cause incorrect behavior. Could you clarify what 'signals used to generate clocks' means and what kind of incorrect behavior is known to occur; are there any specific examples in the repo?

wsnyder, 2023-01-10:
Setting a clock via public changes only the internal value of the signal, it doesn't change how that signal is scheduled, and might also change the signal at a time which may be incorrect in terms of IEEE scheduling semantics.  In v4 this was generally always problematic as the clocks were statically scheduled.  In v5 it's more likely to work OK but no guarantees.

Anyhow the general point there about avoiding --public and using VPI/DPI is still good advice.

Closing as answered, but feel free to follow up if needed.

closed 2023-01-10 by wsnyder

sarutob1, 2023-01-10:
It'd be nice to be able to make just the signals in packages and generally top-level modules public. As a new feature, what would be the preferred way to implement this? I'm thinking of doing it as a scoped annotation similar to lint on...lint off.