verilator/verilator #3850: Optimization Level

MartinSolk, 2023-01-09:
Hello, 
I have a short question which i am not sure about. With the version 4 I compiled System Verilog to C++. When i change the Flag from -Os to -O3, it changed some System Verilog functions from an always comb block to an always sequent block (in C++ comb or sequent function). Because they only triggerd the outputs, this doesn't change the semantic of the code. With Version 5 this doesn't happen. My question is, does the Optimization Flag also affect the Code Generation or is it only later an option for the gcc (or clang) compiler ?

wsnyder, 2023-01-10:
The -O3 level to Verilator currently only changes loop unrolling (and turns on optimizations that are already normally on). This didn't change between v4 and v5, but the scheduling semantics for always blocks did change (hopefully for the more accurate)..

It's not clear from your question if this is causing a problem or not -- closing assuming answers it but feel free to follow up.

closed 2023-01-10 by wsnyder

MartinSolk, 2023-01-10:
Hi,
thanks for the answer. There isn't a problem. Hopefully I posted in the Q/A forum :D
I just wanted to know if the optimization changed something in the conversion.
For example:
I have a Flag which trigger an output like this:

// SYSTEM VERILOG
always_ff@(..)
begin
set OUTPUT1 // Pseudo her
end

always_comb
begin
output1 = (current_state == OUTPUT1) ? 1'b:1'b0;
end

but when i converted it to CPP it changes like this
// CPP
// this is pseudocode know
void sequent1()
{
     set OUTPUT1
     output1 = (current_state == OUTPUT1) ? 1'b:1'b0;
}

As you see the output will be activated the same as in the Verilog, but it changed from a comb block to a sequent block. This didn't happen with the -Os Flag, only when I write -O3. That is why I wondered if the Optimization Flag affect the conversion too. But I am not sure, because I converted something similar in v5 and there it doesn't happen.
Greetings,
Martin

wsnyder, 2023-01-10:
The name of the function in C doesn't necessarily mean it is sequential, it combines functions (V3Combine) and other things that can make that name incorrect.  If it is simulating correctly all should be fine

MartinSolk, 2023-01-10:
Did I understand correctly. It has nothing to do with the optimization level, it could be that the combine functions mixed something up.

wsnyder, 2023-01-10:
The O optimization caused another optimization to be possible that O does not directly control. "mixed something up" implies to me something is wrong, but it is ok.

MartinSolk, 2023-01-10:
Ok thank you for your answers. I think i get it.